dartsim/dart #897: Compilation failure without osg on release-6.1 branch

scpeters, 2017-08-03:
I was recently testing a build of gazebo against a source build of the `release-6.2` branch of dart in our buildfarm, and dart failed to compile with complaints about osg. It works fine with `release-6.1`.

* [release-6.2 failure](https://build.osrfoundation.org/job/gazebo-any_dart4-devel-xenial-amd64-gpu-nvidia_docker/20/console)
* [release-6.1 success](https://build.osrfoundation.org/job/gazebo-any_dart4-devel-xenial-amd64-gpu-nvidia_docker/21/)

During configuration, it fails to find OpenSceneGraph:

~~~
CMake Warning at dart/gui/osg/CMakeLists.txt:30 (message):
  Could NOT find OpenSceneGraph nor OpenThreads -- we will skip dart-gui-osg

  If you believe you do have both OSG and OpenThreads installed, try setting
  OSG_DIR

~~~

but it still tries to compile osg support anyway:

~~~
Scanning dependencies of target dart-gui-osg
[ 86%] Building CXX object dart/gui/osg/CMakeFiles/dart-gui-osg.dir/ImGuiViewer.cpp.o
In file included from /var/lib/jenkins/workspace/gazebo-any_dart4-devel-xenial-amd64-gpu-nvidia_docker/dart/dart/gui/osg/ImGuiViewer.hpp:37:0,
                 from /var/lib/jenkins/workspace/gazebo-any_dart4-devel-xenial-amd64-gpu-nvidia_docker/dart/dart/gui/osg/ImGuiViewer.cpp:32:
/var/lib/jenkins/workspace/gazebo-any_dart4-devel-xenial-amd64-gpu-nvidia_docker/dart/dart/gui/osg/Viewer.hpp:39:28: fatal error: osgViewer/Viewer: No such file or directory
compilation terminated.
dart/gui/osg/CMakeFiles/dart-gui-osg.dir/build.make:62: recipe for target 'dart/gui/osg/CMakeFiles/dart-gui-osg.dir/ImGuiViewer.cpp.o' failed
~~~

mxgrey, 2017-08-03:
After some refactoring was done to DART's CMake setup, we're no longer properly guarding `dart-gui-osg` against the absence of OpenSceneGraph. I have a fix coming up.

jslee02, 2017-08-04:
FYI, this might be an oversight made by #880.

scpeters, 2017-08-04:
I'll race you @mxgrey

mxgrey, 2017-08-04:
It seems I won the race by 2 minutes :sunglasses: 

... although I like your solution better, so I'm stealing it.

mxgrey, 2017-08-07:
Fixed #900 and #898

closed 2017-08-07 by mxgrey