dartsim/dart #917: Proposal: Store headers in a versioned directory

mxgrey, 2017-10-02:
It seems we currently store all dart headers in `include/dart/<namespace>/`, but this makes it difficult to install multiple major versions of dart at the same time.

If we instead stored headers in (as an example) `include/dart<major_version>/dart/<namespace>`, it would be trivial for users to have multiple major versions of dart installed simultaneously. This should only require some tweaks to our current cmake setup, and end users would not notice any change in behavior.

jslee02, 2017-10-02:
:+1: I like this idea.

jslee02, 2017-10-02:
I forgot to mention that this change should be included in the new major version up that @mxgrey might be already aware of.

psigen, 2017-10-02:
What other libraries use this approach?  I'm curious if there are any naming/structure conventions that we can pull from other well known projects.

mxgrey, 2017-10-02:
Right off the top of my head, I know Eigen and Gazebo use this approach.

An alternative solution for installing major versions side-by-side is to make the major version part of the include path name and part of the namespace, like what `tinyxml2` does (e.g. `#include <tinyxml2.h>`), but that makes it very annoying to migrate to new major versions, so I would reject that approach.

jslee02, 2021-04-22:
Addressed by https://github.com/dartsim/dart/pull/1576. The headers are now installed in `include/dartsim/dart<major_version>/dart/<namespace>`. One additional directory depth is added by `dartsim` to avoid polluting the system install directory by multiple directories of `include/dart<major_versions>`.

closed 2021-04-22 by jslee02