dartsim/dart #918: Advice on pull request: simulating realistic human joint limits

jyf588, 2017-10-02:
Hi all,

Karen(@karenliu) and I just finished a DART project using Neural Nets to simulate realistic human range of motion:
https://youtu.be/wzkoE7wCbu0
http://arxiv.org/abs/1709.08685

We were thinking of pushing this tool to DART. I don't think this contribution will affect anything else on current DART. The only potential issue we see is that we use an external neural net library:
https://github.com/tiny-dnn/tiny-dnn

The library is pretty light-weighted, and is header-only & dependency-free. Currently I simply put the source files in a subfolder under dart source, but I am not sure whether this is a good practice.

Another minor thing is that I need to change DART complier dialect to C++14 to get everything work. Will that be a problem?

Could you guys share some insights on this?

Yifeng

jslee02, 2017-10-02:
This sounds awesome! Here are my thoughts:

> The library is pretty light-weighted, and is header-only & dependency-free. Currently I simply put the source files in a subfolder under dart source, but I am not sure whether this is a good practice.

This wouldn't be a problem as `tiny-dnn` suggest that way too. One downside I can imagine now is that it could cause a conflict when using DART with another project that also depends on `tiny-dnn`. If both projects expose the interface of `tiny-dnn` then there would be multiple definition errors. For now, let's keep that way and revisit this issue when it actually becomes problematic.

> Another minor thing is that I need to change DART complier dialect to C++14 to get everything work. Will that be a problem?

I would like to keep DART to be compatible with C++11 since Aikido, a motion planning library that depends on DART, should be compilable on Ubuntu Trusty where Ubuntu Trusty doesn't come with C++14 supporting compiler. Instead, we could make this optional feature only available when DART is built with C++14 compatible compiler.

mxgrey, 2017-10-02:
Depending on how this extension of dart is implemented, we may be able to use `tiny-dnn` internally without conflicts. As long as the `tiny-dnn` headers aren't needed by end-users, we could use them within our source code while compiling and then simply choose not to install them or `#include` them in any `dart` headers.

jyf588, 2017-10-02:
Thanks for the super fast response! In this case, I guess I will first create a pull request of my current implementation, so that you could provide feedback based on the specific code.

closed 2017-11-14 by jyf588