dartsim/dart #920: Move planning component to a separate repository

jslee02, 2017-10-03:
`planning` component will be moved to a separate repository something like `dartsim/dart-planning` as a legacy code repository.

mxgrey, 2017-10-03:
Frankly, I wouldn't mind just deleting it if no one is using it. Moving it to another repository without a dedicated maintainer is essentially a death sentence for it anyway.

If anyone feels sentimental about it (or is actually still using it) and they want to maintain it, then they can fork it off themselves and look after it.

psigen, 2017-10-04:
One reason to branch it might be to allow people who are implementing their
own planners to find it as a reference.

Even if it is deprecated, it might still have use in illustrating how one
might architect a planner. If it's a bad example of that as well, then we
should just delete it.

jslee02, 2017-10-05:
Moving planning component to another repository isn't that tedious work. So let's do it as the consensus here. I believe we all at least agree to move code that isn't maintained by us out from this repository.

PR #923 (2017-10-05): Remove planning component and FLANN dependency
Resolves #920
closed 2017-10-05 by jslee02