dartsim/dart #925: libdart6-collision-ode missing version 6.3

evanglasgow, 2017-10-05:
It looks like libdart6-collision-ode(-dev) is missing a version bump to 6.3. After updating to 6.3 from the PPA I've noticed linker issues with the ODE collision library looking for version 6.2 of DART.

jslee02, 2017-10-05:
Thanks for the report. Which Ubuntu distribution are you on? `libdart6-collision-ode-dev` is available on Xenial and Zesty but Trusty for both of DART 6.2 and 6.3. This is because `libode-dev` on Trusty is too old.

evanglasgow, 2017-10-05:
I'm on Xenial. Strange that I seem to show 6.2 available for libdart6-collision-ode-dev only.

jslee02, 2017-10-05:
I confirmed that it's missing. Let me update here once it's fixed.

jslee02, 2017-10-05:
Fixed! Please close this issue if it works for you as well.

closed 2017-10-06 by evanglasgow