dartsim/dart #927: Consider changing Function::eval[Gradient/Hessian] to const function

jslee02, 2017-10-12:
I believe they're logically correct to be const functions and can't come up with any compelling reason not to do it.

mxgrey, 2017-10-12:
I think in most cases const-qualification will be fine, and normally I aggressively prefer to make function signatures const whenever possible, but the reason I kept them non-const is because they're pure abstract functions, and I don't think we can anticipate how an end-user might want (or need) to implement them. In particular, they might utilize some an object from a third-party library where they need to call non-const functions on the object while the object is a member variable of their `Function` class. Granted there my be ways around this, like marking the member object as mutable, but at the moment I don't see any harm in leaving the functions non-const.

That said, I do agree that logically they should be const-qualified, so I wouldn't really object if you decide to change them.

jslee02, 2017-10-12:
> In particular, they might utilize some an object from a third-party library where they need to call non-const functions on the object while the object is a member variable of their Function class. [...]

Similarly, we should consider the case calling const function in eval functions (non-const function). I think we could also work around this by using `const_cast`. So it would make more sense to make them as const functions, which is logically appropriate and use `mutual` in that special cases. We did this for the getters of Skeleton states with lazy forward kinematics evaluation.

mxgrey, 2017-10-12:
Yeah, I generally agree. My only hesitation is that `const_cast` tends to carry a risk of undefined behavior, which I would rather avoid. And I'm not sure that users will always know about how to properly use the `mutable` keyword.

But admittedly, it may be necessary at some point for a const-qualified object which has a `Function` member variable to be able to invoke the member functions of the `Function`, so I agree: We should const-qualify all of the member functions of `Function`.

PR #928 (2017-10-17): Change Function::eval[Gradient/Hessian] to const functions
Resolves #927

***

**Before creating a pull request**

- [x] Document new methods and classes (nothing new)

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change (nothing new to test)
closed 2017-10-17 by jslee02