dartsim/dart #935: documentation for skel

arpit15, 2017-10-28:
I was wondering if there are some guidelines on what are different elements of .skel file format like .urdf and .sdf. I wanted to create skel file for baxter with other element. However I couldn't find good information on the dart documentation or tutorial page
thank you

mxgrey, 2017-11-07:
This is definitely something that we should consider working on, since the documentation on `.skel` is sparse right now. I think the reason it's been a low priority is because these days the majority of DART users are loading their models with `.urdf` or `.sdf` anyway. I sometimes forget that the `.skel` format even exists.

Is there a particular model feature that you're hoping to express through a `.skel` file which isn't currently possible through a `urdf` or `sdf`? If so, we could use that as motivation for focusing more attention on the `.skel` format. We could also consider expanding `sdf` (or DART's support of `sdf`) to account for what you need.

I think in the long term, we might want to consider phasing out the `.skel` format, since the value in maintaining our own model description format is rather low when there are already a wide range of other formats. The big question is whether `.skel` offers certain features that aren't offered by other (more popular) formats and whether those features can be introduced to those other (more popular) formats.

jslee02, 2017-11-12:
Closing due to lack of a response. Please re-open if you have further trouble.

closed 2017-11-12 by jslee02