dartsim/dart #942: I cannot build the newest dart

Jaemin87, 2017-11-15:
Nice to contact you. My name is Jaemin at UT. 
I just installed the newest dart version on mac. But, when I build, I get some errors as follows: 

[  0%] Building CXX object dart/external/lodepng/CMakeFiles/dart-external-lodepng.dir/lodepng.cpp.o
[  0%] Building CXX object dart/external/imgui/CMakeFiles/dart-external-imgui.dir/imgui.cpp.o
[  0%] Building CXX object dart/external/imgui/CMakeFiles/dart-external-imgui.dir/imgui_draw.cpp.o
[  0%] Building CXX object dart/external/odelcpsolver/CMakeFiles/dart-external-odelcpsolver.dir/error.cpp.o
clangclangclang: : : clang: error: errorerrorno such file or directory: 'NOTFOUND': : errorno such file or directory: 'NOTFOUND'no such file or directory: 'NOTFOUND': no such file or directory: 'NOTFOUND'

In fact, I'm currently using "Apple LLVM version 8.1.0 (clang-802.0.42)". I'm curious that this version may occur the errors that I have. What can I do for solving this issue?

jslee02, 2017-11-15:
It seems there are missing dependencies. Could you run the following command and post the output?
```shell
$ cd <dart_dir>
$ mkdir build; cd build
$ cmake .. -DDART_VERBOSE=On
```

mxgrey, 2017-11-15:
This report might be informative, even if it's a simple case of a missing dependency. If our cmake script is set up correctly, it should catch the fact that a dependency is missing, and then prevent the build from proceeding. This sounds like a required dependency is not being found, but our cmake script is neglecting to throw a `FATAL_ERROR`. Of course it's also possible that @Jaemin87 has a corrupted `CMakeCache.txt` and simply needs to clear the build folder and start over with cmake.

Jaemin87, 2017-11-15:
Thanks for your response. 
Here is the result, it seems there exist the required dependencies excluding OpenMP.

leejaemin ~/dart/build (HEAD) cmake .. -DDART_VERBOSE=On
--
-- ============================================
--                 DART 6.1.2
-- ============================================
--
-- [ Mandatory dependencies for DART core ]
-- Looking for eigen3 - version 3.3.4 found
-- Looking for libccd - version 2.0 found
-- Looking for fcl - version 0.5.0 found
-- Looking for assimp - version 4.0.1 found
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   regex
--   system
--
-- [ Optional dependencies for DART components ]
-- Looking for OpenMP - NOT found
-- Looking for nlopt - version 2.4.2 found
-- Looking for ipopt - version 3.12.8 found
-- Looking for Bullet - found (single precision)
-- Looking for flann - version 1.9.1 found
-- Looking for tinyxml - version 2.6.2 found
-- Looking for tinyxml2 - version 5.0.1 found
-- Looking for urdfdom - 1.0.0 found
-- Looking for OpenGL - version  found
-- Looking for freeglut3 - version  found
-- Looking for OpenSceneGraph - 3.5.7 found
-- Looking for PerlModules - found
-- Looking for doxygen - version 1.8.13 found
--
-- [ Build summary ]
-- CMAKE_GENERATOR  : Unix Makefiles
-- Compiler ID      : Clang
-- Compiler version : 4.2.1

-- Build type       : Release
-- BUILD_SHARED_LIBS: ON
-- ENABLE_OPENMP    : ON
-- Build gui::osg   : ON
-- Install path     : /usr/local
-- CXX_FLAGS        :  NOTFOUND -fPIC -std=c++11 -stdlib=libc++
-- CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_SOURCE_DIR : /Users/leejaemin/dart
-- CMAKE_BINARY_DIR : /Users/leejaemin/dart/build
-- /Users/leejaemin/dart/build/dart/config.hpp
--
-- [ Components ]
-- Adding component: external-imgui
-- Adding component: external-lodepng
-- Adding component: external-odelcpsolver
-- Adding component: dart
-- Adding component: optimizer-ipopt
-- Adding component: optimizer-nlopt
-- Adding component: collision-bullet
-- Adding component: planning
-- Adding component: utils
-- Adding component: utils-urdf
-- Adding component: gui
-- Adding component: gui-osg
--
-- [ Tests ]
-- Adding test: comprehensive/test_Building
-- Adding test: comprehensive/test_Common
-- Adding test: comprehensive/test_Concurrency
-- Adding test: comprehensive/test_Constraint
-- Adding test: comprehensive/test_Frames
-- Adding test: comprehensive/test_InverseKinematics
-- Adding test: comprehensive/test_NameManagement
-- Adding test: comprehensive/test_Distance
-- Adding test: comprehensive/test_Collision
-- Adding test: comprehensive/test_Dynamics
-- Adding test: comprehensive/test_Joints
-- Adding test: comprehensive/test_Skeleton
-- Adding test: comprehensive/test_SoftDynamics
-- Adding test: comprehensive/test_World
-- Adding test: comprehensive/test_ForwardKinematics
-- Adding test: regression/test_Issue000Template
-- Adding test: unit/test_Aspect
-- Adding test: unit/test_ContactConstraint
-- Adding test: unit/test_GenericJoints
-- Adding test: unit/test_Geometry
-- Adding test: unit/test_LocalResourceRetriever
-- Adding test: unit/test_Math
-- Adding test: unit/test_Optimizer
-- Adding test: unit/test_Signal
-- Adding test: unit/test_Subscriptions
-- Adding test: unit/test_Uri
-- Adding test: unit/test_Utilities
-- Adding test: unit/test_CompositeResourceRetriever
-- Adding test: unit/test_FileInfoWorld
-- Adding test: unit/test_PackageResourceRetriever
-- Adding test: unit/test_SdfParser
-- Adding test: unit/test_SkelParser
-- Adding test: unit/test_VskParser
-- Adding test: unit/test_DartLoader
-- Adding test: unit/test_NearestNeighbor
--
-- [ Examples ]
-- Adding example: addDeleteSkels
-- Adding example: atlasSimbicon
-- Adding example: bipedStand
-- Adding example: hardcodedDesign
-- Adding example: hybridDynamics
-- Adding example: jointConstraints
-- Adding example: mixedChain
-- Adding example: operationalSpaceControl
-- Adding example: osgAtlasPuppet
-- Adding example: osgDragAndDrop
-- Adding example: osgEmpty
-- Adding example: osgHuboPuppet
-- Adding example: osgImGui
-- Adding example: osgOperationalSpaceControl
-- Adding example: osgSoftBodies
-- Adding example: osgAtlasSimbicon
-- Adding example: osgTinkertoy
-- Adding example: rigidChain
-- Adding example: rigidCubes
-- Adding example: rigidLoop
-- Adding example: rigidShapes
-- Adding example: simpleFrames
-- Adding example: softBodies
-- Adding example: speedTest
-- Adding example: vehicle
--
-- [ Tutorials ]
-- Adding tutorial: tutorialBiped-Finished
-- Adding tutorial: tutorialBiped
-- Adding tutorial: tutorialCollisions-Finished
-- Adding tutorial: tutorialCollisions
-- Adding tutorial: tutorialDominoes-Finished
-- Adding tutorial: tutorialDominoes
-- Adding tutorial: tutorialMultiPendulum-Finished
-- Adding tutorial: tutorialMultiPendulum
--
-- [ Configured files ]
-- /Users/leejaemin/dart/build/DARTConfig.cmake
-- /Users/leejaemin/dart/build/cmake/DARTConfigVersion.cmake
-- /Users/leejaemin/dart/build/cmake/dart.pc
--
-- Run 'make' to build all the components
-- Run 'make tests' to build all the unittests
-- Run 'make examples' to build all the examples
-- Run 'make tutorials' to build all the tutorials
--
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/leejaemin/dart/build

mxgrey, 2017-11-15:
I would suggest completely deleting your `build` folder and starting over with cmake. If it still fails to build, try running `make VERBOSE=1` and then dump the results into a [gist](https://gist.github.com/) and share a link to the gist here.

Jaemin87, 2017-11-15:
I completely removed `build` folder and started over with cmake. However, I still found error messages when running `make VERBOSE=1` as follows:

`[  0%] Building CXX object dart/external/odelcpsolver/CMakeFiles/dart-external-odelcpsolver.dir/error.cpp.o
cd /Users/leejaemin/dart/build/dart/external/odelcpsolver && /Library/Developer/CommandLineTools/usr/bin/c++  -DBOOST_TEST_DYN_LINK -Ddart_external_odelcpsolver_EXPORTS -I/Users/leejaemin/dart -I/Users/leejaemin/dart/build -I/Users/leejaemin/dart/dart/external/odelcpsolver/dart-external-odelcpsolver -I/Users/leejaemin/dart/dart/external/odelcpsolver/PUBLIC -I/Users/leejaemin/dart/dart/external/odelcpsolver  NOTFOUND -fPIC -std=c++11 -stdlib=libc++ -O3 -DNDEBUG -fPIC   -o CMakeFiles/dart-external-odelcpsolver.dir/error.cpp.o -c /Users/leejaemin/dart/dart/external/odelcpsolver/error.cpp
clang: error: no such file or directory: 'NOTFOUND'
make[2]: *** [dart/external/odelcpsolver/CMakeFiles/dart-external-odelcpsolver.dir/error.cpp.o] Error 1
make[1]: *** [dart/external/odelcpsolver/CMakeFiles/dart-external-odelcpsolver.dir/all] Error 2
make: *** [all] Error 2`

mxgrey, 2017-11-15:
Based on that error message, it looks like there may be an issue with our find module for ODE where it believes that the ODE package has been found, even if it has not actually located ODE's include directories. I'll take a look at our find module for ODE to see if I can confirm this.

Jaemin87, 2017-11-15:
Thanks for your help. I want to let you know that I previously used ODE for Gagebo.

jslee02, 2017-11-15:
This might not be relavent, but it seems you're not building the newest DART according to the cmake output:
```
-- ============================================
--                 DART 6.1.2
-- ============================================
```

Could you make sure the current branch is up to date? Check which branch you're on by running `git branch` and update it by running `git pull`.

Jaemin87, 2017-11-15:
Thanks for your solution. According to the website, I just installed Dart using the command `brew install dartsim6` and then, didn't update the branch (6.1.2 ->7.0.0). That's my mistake. I succeed to build the new Dart and the updated Dart works well for now. Thanks again.

closed 2017-11-18 by jslee02