dartsim/dart #946: Help needed to correct my simple example

anthonyjclark, 2017-12-07:
Hello,

I am working on a simple DART example so that I better learn DART and can compare it to engines that I've used in the past. The example is simply a falling sphere (without graphics). My issue is that I have somehow failed to setup either the joint-body pair and/or the world. **The sphere does not move**.

Could someone help me fix this simple example?

```cpp

#include <dart/dart.hpp>
#include <iostream>

int main()
{
    constexpr double density = 1.0;
    constexpr double radius = 0.3;
    constexpr double restitution = 0.6;
    constexpr double starting_height = 10.0;

    // Create the sphere skeleton (returns a shared_ptr)
    dart::dynamics::SkeletonPtr sphere = dart::dynamics::Skeleton::create();

    // Create joint properties for a free joint that connects the
    // sphere to the world. This is a free joint since the body does
    // not need any constraints with respect to the world.
    dart::dynamics::FreeJoint::Properties sphere_joint_prop;

    // Create body properties for the sphere
    dart::dynamics::BodyNode::Properties sphere_body_prop;

    // Add the sphere body and connect to the world
    std::pair<dart::dynamics::FreeJoint*, dart::dynamics::BodyNode*> sphere_pair =
        sphere->createJointAndBodyNodePair<dart::dynamics::FreeJoint>(
            nullptr, sphere_joint_prop, sphere_body_prop);

    // Create a shape object to attach to the BodyNode (shared_ptr)
    dart::dynamics::ShapePtr sphere_shape(new dart::dynamics::SphereShape(radius));

    // Configure the BodyNode shape so that it has a collision aspect and
    // a dynamic aspect.
    sphere_pair.second->createShapeNodeWith<
        dart::dynamics::CollisionAspect,
        dart::dynamics::DynamicsAspect>(sphere_shape);

    // Inertia
    dart::dynamics::Inertia sphere_inertia;
    const double sphere_mass = density * sphere_shape->getVolume();
    sphere_inertia.setMass(sphere_mass);
    sphere_inertia.setMoment(sphere_shape->computeInertia(sphere_mass));
    sphere_pair.second->setInertia(sphere_inertia);

    // Coefficient of restitution
    sphere_pair.second->setRestitutionCoeff(restitution);

    // Set the initial position of the sphere.
    // The first three components are the orgientation using angle-axis
    // (logmap), the second three components are the translation.
    Eigen::Vector6d positions(Eigen::Vector6d::Zero());
    positions[5] = starting_height;
    sphere_pair.first->setPositions(positions);

    // Create the world object (shared_ptr) and add the sphere skeleton
    dart::simulation::WorldPtr world(new dart::simulation::World);
    world->addSkeleton(sphere);

    // Simulate the world for some amount of time
    constexpr double END_TIME = 5.0;
    while (world->getTime() < END_TIME) {

        std::cout << world->getTime() << " : ";
        for (auto i = 0u; i < sphere->getJoint(0)->getNumDofs(); ++i)
            std::cout << sphere->getJoint(0)->getPosition(i) << " ";
        std::cout << std::endl;

        world->step(false);
    }

    std::cout << world->getGravity() << std::endl;

    return EXIT_SUCCESS;
}
```

```text
# Last several lines of the output
4.991 : 0 0 0 0 0 10
4.992 : 0 0 0 0 0 10
4.993 : 0 0 0 0 0 10
4.994 : 0 0 0 0 0 10
4.995 : 0 0 0 0 0 10
4.996 : 0 0 0 0 0 10
4.997 : 0 0 0 0 0 10
4.998 : 0 0 0 0 0 10
4.999 : 0 0 0 0 0 10
    0
    0
-9.81
```

mxgrey, 2017-12-07:
Could you provide some more information about what platform you're using and what version of DART you're running?

Running on Ubuntu Xenial (16.04) with DART version 6.2.1, I get this result (only taking the last few lines) running your exact code:

```
4.98 : 0 0 0 0 0 -111.67 
4.981 : 0 0 0 0 0 -111.719 
4.982 : 0 0 0 0 0 -111.768 
4.983 : 0 0 0 0 0 -111.817 
4.984 : 0 0 0 0 0 -111.866 
4.985 : 0 0 0 0 0 -111.915 
4.986 : 0 0 0 0 0 -111.964 
4.987 : 0 0 0 0 0 -112.013 
4.988 : 0 0 0 0 0 -112.062 
4.989 : 0 0 0 0 0 -112.111 
4.99 : 0 0 0 0 0 -112.159 
4.991 : 0 0 0 0 0 -112.208 
4.992 : 0 0 0 0 0 -112.257 
4.993 : 0 0 0 0 0 -112.306 
4.994 : 0 0 0 0 0 -112.355 
4.995 : 0 0 0 0 0 -112.404 
4.996 : 0 0 0 0 0 -112.453 
4.997 : 0 0 0 0 0 -112.502 
4.998 : 0 0 0 0 0 -112.551 
4.999 : 0 0 0 0 0 -112.6 
    0
    0
-9.81
```

The full output can be found [here](https://gist.github.com/mxgrey/1d9d4e1565ca9cba32ed423a2ac07e63).

anthonyjclark, 2017-12-07:
I am macOS 10.12.6 (Sierra). I am using DART version 6.3.0

I installed DART with (I imagine I am missing something critical?):

```bash
brew install dartsim6 --without-gui --without-gui-osg --without-optimizer-ipopt --without-optimizer-nlopt --without-planning --without-utils --without-utils-urdf
```

mxgrey, 2017-12-07:
None of what you excluded should have an impact on the core functionality that you're using.

Unfortunately I don't have access to a Mac, so I won't be able to recreate your conditions. Maybe @jslee02 can give it a try when he gets a chance. I've installed 6.3.0 specifically on my Ubuntu machine, and everything still worked as expected, so perhaps this is platform-specific.

This is certainly concerning behavior, and I hope we can find a solid explanation for it.

One thing to consider: Do you have (or have you ever had) multiple versions of DART installed? The header files will not necessarily be compatible between versions (issue #917 has already been opened for this), so it's plausible that mismatching headers could cause unexpected issues. If this is a possible scenario for you, I'd recommend trying to purge DART from your system (I don't know what the process would be for that on Mac) and then reinstall as normal.

mxgrey, 2017-12-08:
If possible, I'd be very interested in seeing the output of the unit tests from your machine. To run the unit tests:

1. clone the source code
2. configure using cmake
3. build the default target using `$ make`
4. build the `tests` target using `$ make tests`
5. run the unit tests using `$ make test`

Our Travis tests are [passing on OSX](https://travis-ci.org/dartsim/dart/jobs/313237847) which apparently [defaults to version 10.11.6](https://docs.travis-ci.com/user/reference/osx/). If the tests fail on your system, that could be informative.

anthonyjclark, 2017-12-08:
I have compile DART from source and run the tests. Here are the results:

```text
Running tests...
Test project /Users/ajc/Documents/projects/adabot02/dart/build
      Start  1: test_Building
 1/22 Test  #1: test_Building ....................   Passed    0.01 sec
      Start  2: test_Common
 2/22 Test  #2: test_Common ......................   Passed    2.37 sec
      Start  3: test_Concurrency
 3/22 Test  #3: test_Concurrency .................   Passed    0.01 sec
      Start  4: test_Constraint
 4/22 Test  #4: test_Constraint ..................   Passed    0.02 sec
      Start  5: test_Frames
 5/22 Test  #5: test_Frames ......................   Passed    0.01 sec
      Start  6: test_InverseKinematics
 6/22 Test  #6: test_InverseKinematics ...........   Passed    0.01 sec
      Start  7: test_NameManagement
 7/22 Test  #7: test_NameManagement ..............   Passed    0.01 sec
      Start  8: test_Distance
 8/22 Test  #8: test_Distance ....................   Passed    0.01 sec
      Start  9: test_Issue000Template
 9/22 Test  #9: test_Issue000Template ............   Passed    0.01 sec
      Start 10: test_Aspect
10/22 Test #10: test_Aspect ......................   Passed    0.01 sec
      Start 11: test_ContactConstraint
11/22 Test #11: test_ContactConstraint ...........   Passed    0.03 sec
      Start 12: test_Factory
12/22 Test #12: test_Factory .....................   Passed    0.01 sec
      Start 13: test_GenericJoints
13/22 Test #13: test_GenericJoints ...............   Passed    0.01 sec
      Start 14: test_Geometry
14/22 Test #14: test_Geometry ....................   Passed    0.01 sec
      Start 15: test_Lemke
15/22 Test #15: test_Lemke .......................   Passed    0.01 sec
      Start 16: test_LocalResourceRetriever
16/22 Test #16: test_LocalResourceRetriever ......   Passed    0.01 sec
      Start 17: test_Math
17/22 Test #17: test_Math ........................   Passed    1.69 sec
      Start 18: test_Optimizer
18/22 Test #18: test_Optimizer ...................   Passed    0.03 sec
      Start 19: test_ScrewJoint
19/22 Test #19: test_ScrewJoint ..................   Passed    0.01 sec
      Start 20: test_Signal
20/22 Test #20: test_Signal ......................   Passed    0.01 sec
      Start 21: test_Subscriptions
21/22 Test #21: test_Subscriptions ...............   Passed    0.01 sec
      Start 22: test_Uri
22/22 Test #22: test_Uri .........................   Passed    0.01 sec

100% tests passed, 0 tests failed out of 22

Total Test time (real) =   4.33 sec
```

mxgrey, 2017-12-08:
Those results are reassuring.

The next thing I would recommend is installing DART from source with `$ sudo make install`, although before you do that, make sure you set the cmake installation prefix so that the installation will replace your homebrew version. On Ubuntu, it would look like:

```
$ cmake .. -DCMAKE_INSTALL_PREFIX=/usr/
$ make -j8
$ sudo make install
```

I'm guessing those commands would also work on OSX, but I have to admit that I don't know for certain.

anthonyjclark, 2017-12-08:
I have run into a problem when trying to do so:

```
cmake -DCMAKE_INSTALL_PREFIX:PATH=/path/to/local/installation ..
make install
```

I can get the falling sphere program to compile, but when running I receive the following:

```
dyld: Library not loaded: @rpath/libdart.6.3.dylib
  Referenced from: /Users/ajc/Documents/projects/adabot02/bin/tmp
  Reason: image not found
```

I'm guessing this is a compiler version/configuration issue?

anthonyjclark, 2017-12-08:
You can ignore my previous comment. It was that the library could not be found due to my installation in a non-standard location (not in `DYLD_LIBRARY_PATH`).

When I run the program using the locally installed DART I get the correct output:

```text
4.991 : 0 0 0 0 0 -112.208
4.992 : 0 0 0 0 0 -112.257
4.993 : 0 0 0 0 0 -112.306
4.994 : 0 0 0 0 0 -112.355
4.995 : 0 0 0 0 0 -112.404
4.996 : 0 0 0 0 0 -112.453
4.997 : 0 0 0 0 0 -112.502
4.998 : 0 0 0 0 0 -112.551
4.999 : 0 0 0 0 0 -112.6
    0
    0
-9.81
```

The question remains though: **why does this not work when I install via homebrew**?

As a test, I uninstalled (`brew uninstall dartsim6`), search my path for any mentions of DART (found none), and reinstalled, which lead to the same issue as before (the sphere does not move).

anthonyjclark, 2017-12-08:
I should also mention that I get the following compilation warning when using this local version (I did not get this warning before):

```bash
/Users/ajc/Documents/projects/adabot02/local/include/dart/collision/DistanceFilter.hpp:45:1: warning: 'DistanceFilter' defined as a struct here but previously
      declared as a class [-Wmismatched-tags]
struct DistanceFilter
^
/Users/ajc/Documents/projects/adabot02/local/include/dart/collision/DistanceOption.hpp:41:1: note: did you mean struct here?
class DistanceFilter;
^~~~~
struct
1 warning generated.
```

mxgrey, 2017-12-08:
That's a great question. I'm relieved that this seems to be a packaging issue rather than a source code bug, but it's still an important issue to resolve.

Hopefully someone who has access to a Mac and/or familiarity with Homebrew will be able to investigate.

jslee02, 2017-12-09:
Let me take a look at this problem with my mac machine when I have a chance.

jslee02, 2017-12-20:
I could reproduce the incorrect result from dart 6.3.0 installed by homebrew. The superficial cause of the problem was that the gravity mode of the body node is set to `false`, which should be `true`. If you add `sphere_pair.second->setGravityMode(true);` after setting the inertia, then the sphere will fall down as expected.

I genuinely don't know why this happens only when installing dart 6.3.0 using homebrew. If I download the tarball ([like the homebrew formulae](https://github.com/dartsim/homebrew-dart/blob/master/Formula/dartsim6.rb#L4)) and build it, the problem doesn't exist.

jslee02, 2017-12-20:
It seems DART 6.3 of Homebrew causes other problems too. Many of AIKIDO's tests segfault with it but not DART 6.3.0 manually built from source.

Please don't use DART 6.3 installed by Homebrew for now. :disappointed: Maybe the formulae should be disabled or replaced by DART 6.2 (if this version works).

closed 2025-11-10 by jslee02