dartsim/dart #948: Help with install instructions

betatim, 2017-12-16:
I am trying to install DART in a docker container as part of https://github.com/openjournals/joss-reviews/issues/500

My [`Dockerfile`](https://github.com/betatim/dart/blob/65dff622d99d3a8763ef1df31f2dc1ca3666cbc8/binder/Dockerfile) is pretty standard I think, and I added the commands from the install instructions however it fails with `apt-add-repository command not found`. Do you know what I need to install to get that command?

jslee02, 2017-12-17:
I'm not an expert on Docker, but [it seems `software-properties-common` needs to be installed on the docker image](https://askubuntu.com/a/38035) as
```dockerfile
# line 3
# Set up locales properly
RUN apt-get update && \
    apt-get install --yes --no-install-recommends locales software-properties-common && \
    apt-get purge && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*
```
Hope this helps.

jslee02, 2018-02-11:
I believe this is resolved. Please let us know if this issue is persistent.

closed 2018-02-11 by jslee02