dartsim/dart #951: Question on setLocalCOM function

ljklonepiece, 2018-01-05:
I am using setLocalCOM to set the location of center of mass of a rigid body with respect to its origin.

I was wondering what mass distribution is used in DART? (unimodal or multimodal or anything else?) 
Given two rigid bodies of the same shapes, masses, and centers of mass, but with different mass distribution, their moments of inertia will be different.

mxgrey, 2018-01-05:
DART does not make any assumptions about mass distribution. You are required to explicitly set the moment of inertia parameters, or load those parameters from a file (e.g. urdf or sdf files).

There are many ways to set the moment of inertia using the DART API. Here are two examples:
```
dart::dynamics::BodyNode* bn = skeleton->getBodyNode("name of body");

// 1
bn->setMomentOfInertia(Ixx, Iyy, Izz, Ixy, Ixz, Iyz);

// 2
dart::dynamics::Inertia inertia = bn->getInertia();
inertia.setMoment(Eigen::Matrix3d( /* ... pass in a moment of inertia matrix ... */ ));
bn->setInertia(inertia);
```

mxgrey, 2018-01-05:
Note that the moment of inertia is assumed to be with respect to the `BodyNode` frame origin.

mxgrey, 2018-01-05:
It may also be worth noting that most of the shape classes (e.g. `dart::dynamics::BoxShape`, dart::dynamics::CylinderShape`) have a built-in static function `computeInertia(~)` for computing a moment of inertia based on a solid geometry assumption. There are a few caveats about these functions:

1. They assume a solid, uniform mass distribution.
2. They compute the moment of inertia with respect to the centroid of the geometry.
3. Be cognizant of the axes of the shape, because those axes have a big influence on the moment of inertia.

The `computeInertia(~)` functions are only there for convenience, to help out in cases where you just want a ballpark moment of inertia and don't care about high fidelity simulation. If you decide to use one of these convenience functions, you'll need to make sure that you rotate and/or shift the moment of inertia matrix based on where your shape is within the frame of the body node. For example, if the shape's center of mass is not at the frame origin, then you'll need to use the parallel axis theorem to shift the moment of inertia. You can find a function called `dart::math::parallelAxisTheorem(~)` in `dart/math/Geometry.hpp` to help with this.

To provide a quick example of how to use this, suppose a body node's geometry is a solid cylinder which has been shifted from the body frame origin (with no change in orientation). You could set that body's mass like so:

```
dart::dynamics::BodyNode* bn = skeleton->getBodyNode("my body node");
const Eigen::Vector3d localComLocation = Eigen::Vector3d(x, y, z); // you provide values for x, y, and z

dart::dynamics::Inertia inertia = bn->getInertia();
inertia.setMass(myMass); // you provide a value for myMass
inertia.setLocalCOM(localComLocation);
const Eigen::Matrix3d I = 
    dart::math::parallelAxisTheorem(
        dart::dynamics::CylinderShape::computeInertia(r, h, myMass), // you provide values for r and h
        localComLocation, myMass);
inertia.setMoment(I);

bn->setInertia(inertia);
```

Disclaimer: I did not test this code before posting it.

I'll go ahead and close this issue. Feel free to reopen it if you have more questions.

closed 2018-01-05 by mxgrey

ljklonepiece, 2018-01-05:
Does the above ways to shift moment of inertia applies to dart::dynamics::MeshShape ?

If I prefer high fidelity simulation,  how should the center of mass and moment of inertia be set?

Thanks a lot!

mxgrey, 2018-01-05:
`dart::dynamics::MeshShape::computeInertia(~)` is by far the least reliable `computeInertia(~)` function, because computing the moment of inertia for an arbitrary shape is not possible with just triangular mesh information. It would only be possible if you assume that the mesh shape is hollow with a very thin, fixed-sized thickness. The `MeshShape::computeInertia(~)` function that we provide is extremely crude and only computes the moment of inertia of an axis-aligned bounding box around the mesh. This is a terribly inaccurate estimate, and should only be used for small, simple, squarish meshes as a first-order approximation for cases where convenience is much more important than accuracy.

I should also note that `MeshShape::computeInertia(~)` does not provide a static version like the other shape types. To use it, you must create a `MeshShape` object and then call the function on that instance.

Getting an accurate model of a body's moment of inertia is a field of research in-and-of itself, called System Identification. For the highest fidelity simulation possible, you should take a real-world version of the object and run experiments to measure its moment of inertia parameters. But since that's a very difficult and time-consuming thing to do, your best bet is probably to just use CAD software like SolidWorks, Autodesk Inventor, or AutoCAD and have one of those compute an estimate of the moment of inertia.