dartsim/dart #952: CMakeLists.txt in example/

junhyeokahn, 2018-01-05:
Hi,
Thanks for your efforts to maintain this nice dynamics simulator.

I successfully install Dart from source (Dart 7.0.0, mac os sierra)
  - Install dependancies through homebrew
  - clone this repository
  - I made build directory and did ```cmake ..``` and ```make examples```
  - Therefore, I could run the example through ```../osgOperationalSpaceControl```
Next I would like to install dart in my local and use it.
  - So I did ```sudo make install```.
  - In examples/operationalSpaceControl directory, made build directory.
  - However, I got an error as below due to dart version.
```
-- The C compiler identification is AppleClang 9.0.0.9000039
-- The CXX compiler identification is AppleClang 9.0.0.9000039
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:5 (find_package):
  Could not find a configuration file for package "DART" that is compatible
  with requested version "6.2.0".

  The following configuration files were considered but not accepted:

    /usr/local/share/dart/cmake/DARTConfig.cmake, version: 7.0.0

-- Configuring incomplete, errors occurred!
```
  - In order to fix it, I've changed CMakeLists.txt in operationalSpaceControl directory and used
```
find_package(dart COMPONENTS utils-urdf gui CONFIG)
```
instead of
```
find_package(DART 6.2.0 REQUIRED COMPONENTS utils-urdf gui CONFIG)
```
-- I was able to do ```cmake ..``` successfully but when I ```make```, I got an error as below.
```
Scanning dependencies of target operationalSpaceControl
[ 25%] Building CXX object CMakeFiles/operationalSpaceControl.dir/Controller.cpp.o
[ 50%] Building CXX object CMakeFiles/operationalSpaceControl.dir/Main.cpp.o
[ 75%] Building CXX object CMakeFiles/operationalSpaceControl.dir/MyWindow.cpp.o
/Users/junhyeok/Repository/dart/examples/operationalSpaceControl/MyWindow.cpp:142:3: warning: 'glutPostRedisplay' is deprecated:
      first deprecated in macOS 10.9 [-Wdeprecated-declarations]
  glutPostRedisplay();
  ^
/System/Library/Frameworks/Glut.framework/Headers/glut.h:444:22: note: 'glutPostRedisplay' has been explicitly marked deprecated here
extern void APIENTRY glutPostRedisplay(void) OPENGL_DEPRECATED(10_0, 10_9);
                     ^
/Users/junhyeok/Repository/dart/examples/operationalSpaceControl/Main.cpp:62:3: warning: 'glutInit' is deprecated: first deprecated
      in macOS 10.9 [-Wdeprecated-declarations]
  glutInit(&argc, argv);
  ^
/System/Library/Frameworks/Glut.framework/Headers/glut.h:431:22: note: 'glutInit' has been explicitly marked deprecated here
extern void APIENTRY glutInit(int *argcp, char **argv) OPENGL_DEPRECATED(10_0, 10_9);
                     ^
/Users/junhyeok/Repository/dart/examples/operationalSpaceControl/Main.cpp:64:3: warning: 'glutMainLoop' is deprecated: first
      deprecated in macOS 10.9 [-Wdeprecated-declarations]
  glutMainLoop();
  ^
/System/Library/Frameworks/Glut.framework/Headers/glut.h:438:22: note: 'glutMainLoop' has been explicitly marked deprecated here
extern void APIENTRY glutMainLoop(void) OPENGL_DEPRECATED(10_0, 10_9);
                     ^
1 warning generated.
2 warnings generated.
[100%] Linking CXX executable operationalSpaceControl
Undefined symbols for architecture x86_64:
  "dart::io::DartLoader::parseSkeleton(dart::common::Uri const&, std::__1::shared_ptr<dart::common::ResourceRetriever> const&)", referenced from:
      _main in Main.cpp.o
  "dart::io::DartLoader::DartLoader()", referenced from:
      _main in Main.cpp.o
  "dart::gui::GlutWindow::screenshot()", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::GlutWindow::move(int, int)", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::GlutWindow::specKey(int, int, int)", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::GlutWindow::simTimer(int)", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::Win3D::initLights()", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::Win3D::initWindow(int, int, char const*)", referenced from:
      _main in Main.cpp.o
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::Win3D::drag(int, int)", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::Win3D::click(int, int, int, int)", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::Win3D::initGL()", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::Win3D::render()", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::Win3D::resize(int, int)", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::displayTimer(int)", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawEntities()", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::draw()", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::keyboard(unsigned char, int, int)", referenced from:
      MyWindow::keyboard(unsigned char, int, int) in MyWindow.cpp.o
  "dart::gui::SimWindow::setWorld(std::__1::shared_ptr<dart::simulation::World>)", referenced from:
      _main in Main.cpp.o
  "dart::gui::SimWindow::drawSkels()", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::SimWindow()", referenced from:
      MyWindow::MyWindow(Controller*) in MyWindow.cpp.o
  "dart::gui::SimWindow::~SimWindow()", referenced from:
      MyWindow::MyWindow(Controller*) in MyWindow.cpp.o
      MyWindow::~MyWindow() in MyWindow.cpp.o
  "dart::gui::SimWindow::drawEntity(dart::dynamics::Entity const*, Eigen::Matrix<double, 4, 1, 0, 4, 1> const&, bool) const", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawMarker(dart::dynamics::Marker const*, Eigen::Matrix<double, 4, 1, 0, 4, 1> const&, bool) const", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawBodyNode(dart::dynamics::BodyNode const*, Eigen::Matrix<double, 4, 1, 0, 4, 1> const&, bool, bool) const", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawSkeleton(dart::dynamics::Skeleton const*, Eigen::Matrix<double, 4, 1, 0, 4, 1> const&, bool) const", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawSkeletons() const", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawShapeFrame(dart::dynamics::ShapeFrame const*, Eigen::Matrix<double, 4, 1, 0, 4, 1> const&, bool) const", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawPointMasses(std::__1::vector<dart::dynamics::PointMass*, std::__1::allocator<dart::dynamics::PointMass*> >, Eigen::Matrix<double, 4, 1, 0, 4, 1> const&, bool) const", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawShape(dart::dynamics::Shape const*, Eigen::Matrix<double, 4, 1, 0, 4, 1> const&) const", referenced from:
      vtable for MyWindow in MyWindow.cpp.o
  "dart::gui::SimWindow::drawWorld() const", referenced from:
      MyWindow::drawWorld() const in MyWindow.cpp.o
  "typeinfo for dart::gui::SimWindow", referenced from:
      typeinfo for MyWindow in MyWindow.cpp.o
  "_glutInit", referenced from:
      _main in Main.cpp.o
  "_glutMainLoop", referenced from:
      _main in Main.cpp.o
  "_glutPostRedisplay", referenced from:
      MyWindow::keyboard(unsigned char, int, int) in MyWindow.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [operationalSpaceControl] Error 1
make[1]: *** [CMakeFiles/operationalSpaceControl.dir/all] Error 2
make: *** [all] Error 2
```
-- Other examples also show this behaviors. I guess this current cmakefile in the repository is not correctly working.
I would really appreciate if someone could help me.
Thank you.

jslee02, 2018-01-05:
Thanks for the detailed description. I believe the error caused because the major version up (6 --> 7), and `find_package` complains about it. #953 should fix this issue.

junhyeokahn, 2018-01-05:
Thanks for the reply.
One thing to add is that I should changed package names as well as dart version. (Some of name is changed. I checked it in /usr/local/share/dart/cmake.
Finally I used below line instead and it works.
```
cmake_minimum_required(VERSION 2.8.12)

project(operationalSpaceControl)

find_package(DART 7.0.0 REQUIRED COMPONENTS io-urdf gui CONFIG)

add_compile_options(-std=c++11)

include_directories(${DART_INCLUDE_DIRS})

file(GLOB srcs "*.cpp" "*.hpp")
add_executable(${PROJECT_NAME} ${srcs})
target_link_libraries(${PROJECT_NAME} ${DART_LIBRARIES})
```
Thank you!

closed 2018-01-05 by junhyeokahn

jslee02, 2018-01-05:
Ah, forgot to mention that. 😅  That name change is also included in #953.

areebm, 2018-03-14:
What package names were changed in .../dart/cmake? Performing the edits to the CMakeLists file did not resolve this issue for me. I've installed DART v6.3.0

mxgrey, 2018-03-14:
If you're using `v6.3.0`, it will still use `utils` instead of `io`.

areebm, 2018-03-19:
In fact I did try `utils` instead of `io`, but it was still giving errors. I resolved to use `make -j4 ...` to compile, as [shown](https://dartsim.github.io/install_dart_on_ubuntu.html), for the tutorials I am working on.
**Note***: I think the [tutorial instructions](https://dartsim.github.io/tutorials_pendulum.html#lesson-1a-reset-everything-to-default-appearance) are using deprecated code which is not included in the finished tutorial cpp files.

jslee02, 2018-03-19:
Just to clarify, are you getting the same error messages as @junhyeokahn got? If not, it would be helpful if you share the error messages. Also, which branch or tag of DART are you compiling?

areebm, 2018-03-20:
I am compiling the most recent, stable tag of Dart 6, `v6.3.0`. Using `cmake` instead of `make j4 ...` I was receiving this error:
```
CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a package configuration file provided by "DART" (requested
  version 6.3.0) with any of the following names:

    DARTConfig.cmake
    dart-config.cmake

  Add the installation prefix of "DART" to CMAKE_PREFIX_PATH or set
  "DART_DIR" to a directory containing one of the above files.  If "DART"
  provides a separate development package or SDK, be sure it has been
  installed.
```
I added this line per the instructions in the above error
`set(DART_DIR /home/dart/build)`
 
And I received the error similar to @junhyeokahn's
  ```
Could not find a configuration file for package "DART" that is compatible
  with requested version "6.3.0".

The following configuration files were considered but not accepted:

    /home/dart/build/DARTConfig.cmake, version: unknown
```

mxgrey, 2018-03-20:
That's cmake's way of telling you that you need to install DART before you can build the example on its own.

You can build the example without installing DART by running cmake on the top-level CMakeLists.txt and then calling `make examples`.

You cannot point `DART_DIR` at a build directory, because a build directory is not the same as an install directory.

Also, you should not hard code a path in a `CMakeLists.txt` like that. Instead of saying `set(DART_DIR /home/dart/build)`, you should add the argument `-DDART_DIR=/home/dart/build` when you call the `cmake` command. cmake arguments that look like `-D<variable>=<value>` will be effectively the same as putting `set(<variable> <value>)` into your CMakeLists.txt, except you don't need to change the script if you move the directory somewhere.

jslee02, 2018-03-20:
> I've installed DART v6.3.0

Assuming you've installed DART, could you run `find /usr/ -name DARTConfig.cmake` and post the output?

areebm, 2018-03-21:
I inherited this project from someone else who said that DART had been installed on the system. However, `find /usr/ -name DARTConfig.cmake` returned nothing until after I went back and installed DART myself:
`/usr/local/share/dart/cmake/DARTConfig.cmake`
 cmake is also working now. Thanks for the help!

PR #953 (2018-01-05): Update CMake scripts of examples/tutorials for DART 7
Resolves #952 

***

**Before creating a pull request**

- [x] Document new methods and classes (N/A)

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change (N/A)