dartsim/dart #954: Clang 5.0.0 warning per -Winjected-class-name

jslee02, 2018-01-05:
### Bug Report
> Please answer the following questions for yourself before submitting an issue.
- [x] I checked the [documentation](http://dartsim.github.io/) and found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
* Select DART version: master (but it might be applied to any version that contains `Aspect` class)
* Select OS name and version name(or number): Ubuntu 14.04
* Select compiler name and version number: Clang 5.0.0

#### Expected Behavior
Compiled w/o any warnings

#### Current Behavior
[Travis CI complains](https://travis-ci.org/dartsim/dart/jobs/325463914#L1726) as
```
/home/travis/build/dartsim/dart/dart/common/detail/CompositeData.hpp:56:34: error: ISO C++ specifies that qualified reference to 'Aspect' is a constructor name rather than a type in this context, despite preceding 'typename' keyword [-Werror,-Winjected-class-name]
      AspectT, typename AspectT::Aspect>::type;
```

mxgrey, 2018-01-05:
I've made some progress on digesting the error message. I'm a tiny bit concerned this might be a compiler bug, because I don't quite see how what it's saying can be true.

Based on the printout, it seems that `AspectT` in this case is `dart::common::EmbeddedPropertiesAspect<dart::dynamics::EntityNode<dart::dynamics::TemplatedJacobianNode<dart::dynamics::FixedJacobianNode> >, dart::dynamics::detail::EntityNodeProperties>`, which gets created by `EmbedProperties<...>` when `EmbedPropertiesOnTopOf<...>` is inherited by `EndEffector`.

What's strange is that this `AspectT` inherits `common::CompositeTrackingAspect<dynamics::EntityNode<...>>` which in turn inherits `common::Aspect`. Since `AspectT` inherits `Aspect`, we should see that `std::is_base_of<Aspect, AspectT>::value` evaluates to `true`, and therefore `typename AspectT::Aspect` should never be compiled in this context. As far as I understand the ISO rules, `typename AspectT::Aspect` should be completely ignored in this context.

Instead, it seems like `std::is_base_of<Aspect, AspectT>` is incorrectly evaluating to `false` here, or else `std::conditional<true, A, B>::type` is evaluating the expression `B` when I would think it's not supposed to.

I'll see about creating a minimal working example of this failure. That should make it easier to identify whether this is Clang misbehaving (admittedly unlikely) or whether a recent change to the ISO standard is going to make the implementation of this needlessly difficult.

jslee02, 2018-02-12:
I believe this is resolved by #963

closed 2018-02-12 by jslee02