dartsim/dart #957: Add publications to DART's Academic Research page

costashatz, 2018-01-15:
Hello,

I was checking for something in your docs and found the *Academic Research* page (I wasn't aware of this doc page): http://dartsim.github.io/use_cases.html

There you have quite a few of publications that use DART. In our research lab (http://www.resibots.eu/), we have used DART in the following publications:

**Peer-reviewed International Journals and Conferences**

- Chatzilygeroudis, K., & Mouret, J. B. (2018). [Using Parameterized Black-Box Priors to Scale Up Model-Based Policy Search for Robotics](https://arxiv.org/pdf/1709.06917). *Proceedings of the International Conference on Robotics and Automation (ICRA)*.
- Pautrat, R., Chatzilygeroudis, K., & Mouret, J.-B. (2018). [Bayesian Optimization with Automatic Prior Selection for Data-Efficient Direct Policy Search](https://arxiv.org/pdf/1709.06919). *Proceedings of the International Conference on Robotics and Automation (ICRA)*.
- Supratik P., Chatzilygeroudis K., Ciosek K., Mouret JB., Osborne M., and Whiteson Sh. (2018). [Alternating Optimisation and Quadrature for Robust Control](http://www.cs.ox.ac.uk/people/shimon.whiteson/pubs/paulaaai18.pdf). *Proceedings of AAAI Conference on Artificial Intelligence*.
- Chatzilygeroudis, K., Vassiliades, V. and Mouret, J.-B. (2017). [Reset-free Trial-and-Error Learning for Robot Damage Recovery](https://arxiv.org/pdf/1610.04213). *Robotics and Autonomous Systems*.
- Vassiliades V., Chatzilygeroudis K., and Mouret JB. (2016). [Using centroidal voronoi tessellations to scale up the multi-dimensional archive of phenotypic elites algorithm](https://arxiv.org/pdf/1610.05729.pdf). *IEEE Transactions on Evolutionary Computation*.

**Minimally reviewed ones (workshops)**

- Chatzilygeroudis, K., Cully, A. and Mouret, J.-B. (2016). [Towards semi-episodic learning for robot damage recovery](https://arxiv.org/abs/1610.01407). *Workshop on AI for Long-Term Autonomy at the IEEE International Conference on Robotics and Automation 2016*.
- Papaspyros, V., Chatzilygeroudis, K., Vassiliades, V., and Mouret, J.-B. (2016). [Safety-Aware Robot Damage Recovery Using Constrained Bayesian Optimization and Simulated Priors](https://arxiv.org/pdf/1611.09419v3). *Workshop on Bayesian Optimization at the Annual Conference on Neural Information Processing Systems (NIPS) 2016.*

It'd be nice to include (some of) them in the DART docs.

Thanks in advance!

jslee02, 2018-01-16:
Thanks for introducing publications using DART. Awesome! 

I added three of them to the [website](http://dartsim.github.io/use_cases.html) because I couldn't find that DART is cited in the other four papers. :sweat:

costashatz, 2018-01-16:
> I added three of them to the website because I couldn't find that DART is cited in the other four papers.

That would be by mistake.. I will update the publications and let you know (I will also give you some pictures to put :smile: )..

costashatz, 2018-11-23:
Hey,

Here's the updated list of publications (now they should contain DART in their citations) with images:

There you have quite a few of publications that use DART. In our research lab (http://www.resibots.eu/), we have used DART in the following publications:

**Peer-reviewed International Journals and Conferences**

- Chatzilygeroudis, K., & Mouret, J. B. (2018). [Using Parameterized Black-Box Priors to Scale Up Model-Based Policy Search for Robotics](https://arxiv.org/pdf/1709.06917). *Proceedings of the International Conference on Robotics and Automation (ICRA)*.
[image]

- Pautrat, R., Chatzilygeroudis, K., & Mouret, J.-B. (2018). [Bayesian Optimization with Automatic Prior Selection for Data-Efficient Direct Policy Search](https://arxiv.org/pdf/1709.06919). *Proceedings of the International Conference on Robotics and Automation (ICRA)*.
[image]

- Supratik P., Chatzilygeroudis K., Ciosek K., Mouret JB., Osborne M., and Whiteson Sh. (2018). [Alternating Optimisation and Quadrature for Robust Control](https://arxiv.org/pdf/1605.07496.pdf). *Proceedings of AAAI Conference on Artificial Intelligence*.
[image]

- Chatzilygeroudis, K., Vassiliades, V. and Mouret, J.-B. (2017). [Reset-free Trial-and-Error Learning for Robot Damage Recovery](https://arxiv.org/pdf/1610.04213). *Robotics and Autonomous Systems*.
[image]

- Vassiliades V., Chatzilygeroudis K., and Mouret JB. (2016). [Using centroidal voronoi tessellations to scale up the multi-dimensional archive of phenotypic elites algorithm](https://arxiv.org/pdf/1610.05729.pdf). *IEEE Transactions on Evolutionary Computation*.
[image]

By the way, maybe you could add our DART wrapper to the software use cases:

[robot_dart](https://github.com/resibots/robot_dart): *A generic and lightweight wrapper over DART simulator for fast and flexible robot simulations*.

jslee02, 2018-12-16:
@costashatz I updated the [webpage](http://dartsim.github.io/use_cases.html) correspondingly. Thank you for sharing the great work and the updates!

Closing this issue but feel free to reopen if you would like to add more.

closed 2018-12-16 by jslee02