dartsim/dart #961: FCL is too slow?

costashatz, 2018-01-24:
Hello,

#### Environment
* Select DART version: master and 6.3.0
* Select OS name and version name: Ubuntu 16.04 
* Select compiler name and version number: GCC 5.4.0
* FCL version: tried both 0.3.2 and 0.5.0

#### Current Behavior
I am using the dominoes tutorial and I create several of them (like 8-12), I push the first domino and the simulation is really slow when the collisions start to happen (i.e., really really bad)! I altered a bit the tutorial to not include the arm (which has meshes for collision) and switch to the simple `DARTCollisionDetector` and now I can put much much more dominoes (throw them from random heights/positions with collisions etc) and the simulation is really smooth. I tried the default FCL in my machine (0.3.2) and the latest (stable) version (0.5.0), but nothing changed.

Is this expected? Or am I missing something (for example, ways of making the collision checking faster!)?

Thanks!

jslee02, 2018-01-26:
This is somewhat an expected result. There is two main reasons behind this: (1) collision handling method in DART and (2) number of contact points generated by the collision detector.

(1) DART uses rather slow but more precise contact constraint solver. 

AFAIK, there are two major categories of solving non-penetration contact constraint handling methods: (a) [LCP](https://en.wikipedia.org/wiki/Linear_complementarity_problem)-based method and (b) iterative method. I believe DART initially chose (a) method to compute precise contact points sacrificing speed, but we are working on supporting (b) for interactive simulations like manipulation in VR. [This](https://github.com/dartsim/dart/tree/enhancement/constraint_solver) is my working branch hoping to finish initial implementation by next couple of months. 

(2) `FCLCollisionDetector` generally returns more contact points than `DARTCollisionDetector`.

In FCL setting of DART, we use meshes even for primitive shapes (e.g., box and sphere) by default, and this generally results in more contact points than primitive-primitive shape pair contacts (like `DARTCollisionDetector`). This is because FCL detects all the contact points of the meshes' triangle pairs in contact while primitive-primitive collision algorithm returns minimally required contacts (this can be varied by the concrete primitive collision algorithm). For example, box-box collision algorithm returns up to four contacts.

Loosely speaking, the complexity of (a) method is O(n^3) where n is the number of contacts (assuming no friction). That said, more contacts make the simulate slow quickly.

The reason we don't use FCL's primitive shape support is that it has a few unresolved problems. I had worked on addressing those issues (e.g., https://github.com/flexible-collision-library/fcl/pull/52, https://github.com/flexible-collision-library/fcl/pull/61), but there are some remains.

There is persistent contact manifold technic of Bullet Physics engine that can address this issue. It manages the contact points reported from the collision detector to be less than 4, which is minimal contact points to support the contact shapes each other. I'm also considering having the similar feature in DART.

jslee02, 2018-02-12:
Closing. Please reopen if you have a further issue. :)

closed 2018-02-12 by jslee02