dartsim/dart #965: urdf load with materials

junhyeokahn, 2018-01-31:
Hello,
I am trying to import valkyrie model from urdf with materials.
I follow the tutorial and have a setcolormode for each body node if it has one.
```
void setMeshColor(dart::dynamics::SkeletonPtr robot) {
    for(std::size_t i=0; i<robot->getNumBodyNodes(); ++i) {
        dart::dynamics::BodyNode* bn = robot->getBodyNode(i);
        auto shapeNodes = bn->getShapeNodesWith<dart::dynamics::VisualAspect>();
        for(auto shapeNode : shapeNodes) {
            std::shared_ptr<dart::dynamics::MeshShape> mesh =
                std::dynamic_pointer_cast<dart::dynamics::MeshShape>(
                        shapeNode->getShape());
            if(mesh)
                mesh->setColorMode(dart::dynamics::MeshShape::SHAPE_COLOR);
        }
    }
}
```
However, I could only see the gray or blue robot depends on color mode.
It seems like it doesn't find the material even though correct path is written in 'dae' file.

To make it easy I made a repository with urdf and models.
1. git clone https://github.com/junhyeokahn/dartDebug.git
2. At the repository, ```mkdir build && cd build```
3. ```cmake ..```
4. ```build -j```
5. ```./valkyrie```

Plus, is there any way to zoom in and out in osg? The instruction says scroll the wheel but it does not work.

I apologize for the messy description and always thanks for your effort.

mxgrey, 2018-02-01:
Note that the example expects DART version 7.0, which is currently only available by compiling the master branch from source (as far as I know).

Setting the mesh color mode to `SHAPE_COLOR` will use whatever color you pass to the `ShapeNode` that holds the shape, using  `ShapeNode::getVisualAspect()->setColor(~)`, so that mode will definitely hide the material color. You'll definitely want to use `MATERIAL_COLOR`.

When I run your code using `MATERIAL_COLOR`, it comes up as a shiny gray color (although the apparent shininess might just be caused by the geometry's curvature). I take it this is the "gray" you were referring to, which means this is not the expected material color.

There are three places where the material color might be getting lost:

1. Assimp is parsing the `.dae` incorrectly

2. `dart-gui-osg` is reading the material from assimp incorrectly

3. `dart-gui-osg` is sending the material to `osg` incorrectly.

It's difficult to guess which of those it might be. The best course of action would probably be whittling down the possible causes.

If there is another application somewhere that can load and display a `.dae` with assimp, then you could load the model in that application and see if the materials are correct.

We could try loading the model in the older OpenGL `dart-gui` library and see if it handles the material properties correctly.

If both of those work, then we'll have narrowed it down to an issue in `dart-gui-osg`, which would be very informative.

> Plus, is there any way to zoom in and out in osg? The instruction says scroll the wheel but it does not work.

Could you let us know what platform you're running on? Using Ubuntu 16.04, I had no problem using the scroll wheel to zoom in and out in the OSG window that your example produces.

junhyeokahn, 2018-02-01:
Thanks for the reply. For zoom, I am currently using Mac OS Sierra.
I just realized attached dae file is loaded correctly and the difference is previous valkyrie dae file is pointing png file but attached dae file includes color information inside it.
[leg.dae.zip](https://github.com/dartsim/dart/files/1683830/leg.dae.zip)
Does dart support to read png file?

jslee02, 2018-02-01:
> Does dart support to read png file?

Currently, none of OpenGL and OSG renderers of DART load texture images (only texture coordinates are parsed by the OSG renderer). I think we need to have something like [this](https://github.com/openscenegraph/OpenSceneGraph/blob/master/examples/osgtexture2D/osgtexture2D.cpp#L170-L176) to load texture images. I confirmed this works.

[image]

Please note that there are several holes on the pelvis mesh, which are loaded from `pelvistexture.png`.

Let me submit a PR for this within a couple of days.

junhyeokahn, 2018-02-01:
I appreciate it :)

jslee02, 2018-02-05:
I've made some progress on this. Here is an Valkyrie example with texture images:
[image]

A PR will be created for this soon.

junhyeokahn, 2018-02-05:
Wow, it's awesome :). Thank you so much.

jslee02, 2018-02-13:
I believe this is addressed by #973.

PR #973 (2018-02-13): Render texture images when available
Before:
[image]

After:
[image]

Depends on #972 
Resolves #965 

***

**Before creating a pull request**

- [ ] Document new methods and classes

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [ ] Add unit test(s) for this change
closed 2018-02-13 by jslee02