dartsim/dart #969: Displaying local frame of link, joint, and collision

dhkim0821, 2018-02-04:
Is there an easy way to display the local frames of links, joints, and collisions in GUI?

mxgrey, 2018-02-04:
If you use the `dart-gui-osg` (OpenSceneGraph-based) library, you can use the `dart::gui::osg::InteractiveFrame` class.

Sample code would look something like this (warning: I have not tested this code):

```
#include <dart/gui/osg/InteractiveFrame.hh>

/* ... other things ...*/

void displayAllFrames(
    const dart::simulation::WorldPtr& world, 
    const dart::dynamics::ConstSkeletonPtr& robot)
{
  for(std::size_t i=0; i < robot->getNumBodyNodes(); ++i)
  {
    const BodyNode* bn = robot->getBodyNode(i);
    world->addSimpleFrame(std::make_shared<dart::gui::osg::InteractiveFrame>(
        bn, bn->getName()+"/frame"));
    for(std::size_t j=0; j < bn->getNumChildJoints(); ++j)
    {
      const Joint* joint = bn->getChildJoint(j);
      const Eigen::Isometry3d offset = joint->getTransformFromParentBodyNode();
      world->addSimpleFrame(std::make_shared<dart::gui::osg::InteractiveFrame>(
          bn, joint->getName()+"/frame"), offset));
    }
  }
}

/* ... other things ... */
```

The `InteractiveFrame` class is meant to be used as a drag-and-drop marker, but if you just add it as a `SimpleFrame` to the `world` and don't enable the drag-and-drop feature for it, then it makes a good visual for a generic frame.

By default, it contains three axes, three planes, and three circles, but you can toggle off any of those using the `getTool()` function. Here's an example of creating an interactive frame, and then turning off the circles and the planes (again, this is untested code, off the top of my head):

```
dart:;gui::osg::InteractiveFramePtr frame = 
    std::make_shared<dart::gui::osg::InteractiveFrame>(dart::dynamics::Frame::World());
for(const auto type : {dart::gui::osg::InteractiveTool::ANGULAR, dart::gui::osg::InteractiveTool::PLANAR})
  for(std::size_t i=0; i < 3; ++i)
    frame->getTool(type, i)->setEnabled(false);
```

I think that the the `dart::gui::osg::InteractiveFrame` class could also be used in the OpenGL `dart::gui` library since it doesn't really have any dependencies on `osg`. But our current code structure requires you to have `OpenSceneGraph` installed before you can access the `InteractiveFrame` class. We should probably consider migrating `dart::gui::osg::InteractiveFrame` to `dart::gui::InteractiveFrame`.

jslee02, 2018-02-04:
@dhkim0821 Make sure you set `size_scale` of `InteractiveFrame` sufficiently larger than the visual object so that you can see the axes out of the visual object.

@mxgrey How Gazebo render the axes? Does it also have the issue of not visible when the visual object is too large or automatically adjust the size of axes? :thinking:

mxgrey, 2018-02-04:
In Gazebo, there's a [LinkFrameVisual](https://bitbucket.org/osrf/gazebo/src/8a59b5f0f0b9bda21ee905098b47900da492ba9c/gazebo/rendering/LinkFrameVisual.cc?at=default&fileviewer=file-view-default) which has a function that performs a scaling based on the bounding box of the link that it's attached to. I would guess that function needs to be manually called any time the link geometry gets changed, unless the client Gazebo application is monitoring changes to link geometry sizes.

There's also a separate [JointVisual](https://bitbucket.org/osrf/gazebo/src/8a59b5f0f0b9bda21ee905098b47900da492ba9c/gazebo/rendering/JointVisual.hh?at=default&fileviewer=file-view-default) class that customizes its behavior per joint type.

dhkim0821, 2018-02-05:
The code is working! The small change I made is adding type casting in front of body node pointer:
world->addSimpleFrame(std::make_shared<dart::gui::osg::InteractiveFrame>((dart::dynamics::Frame*)bn, bn->getName()+"/frame"));

[image]

mxgrey, 2018-02-05:
Ah, I forgot that you can't pass in a `const Frame*` to the `InteractiveFrame` constructor.

Instead of casting it, it would be more appropriate to change

```
const dart::dynamics::ConstSkeletonPtr& robot)
```
to
```
const dart::dynamics::SkeletonPtr& robot)
```

and then
```
const BodyNode* bn = robot->getBodyNode(i);
```
to
```
BodyNode* bn = robot->getBodyNode(i)
```

That would be the "const-correct" way to fix the code.

I'll go ahead and close this issue since it seems to be resolved.

closed 2018-02-05 by mxgrey

lisheng-k, 2021-09-30:
> @dhkim0821 Make sure you set `size_scale` of `InteractiveFrame` sufficiently larger than the visual object so that you can see the axes out of the visual object.
> 
> @mxgrey How does Gazebo render the axes? Does it also have the issue of not visible when the visual object is too large or automatically adjust the size of axes? thinking

Hi,

I am also trying to use add the interaqctiveframe onto the objects, and I did that with the same method shown above. 

but the frame is too big in scale (visual) respective to the objects, the dimension of the objects are in scale around, 5 ~ 10 cm.

so with that objects and the frames, all look like a mess.

I am wandering is there a method or function which can scale down the frame into mm or at least cm?

thanks in advance.
[image]