dartsim/dart #971: Displaying shadow

dhkim0821, 2018-02-05:
Is there any way to add shadow in DART?

mxgrey, 2018-02-05:
DART does not inherently provide any features for rendering shadows, and I don't believe there are any plans to do so, ever, since DART is first and foremost a dynamics toolkit. Graphical shading falls far outside the scope of DART.

That being said, OpenSceneGraph does provide features for rendering shadows, and the `dart-gui-osg` wrapper gives a significant amount of access into the underlying OpenSceneGraph objects that are used for rendering.

To get shadows in a DART gui application, I recommend looking into how to achieve that through native OpenSceneGraph features. I have not tried doing that yet, so if there is anything about the current design of `dart-gui-osg` that makes it prohibitively difficult, please let us know, and we can work out a redesign of the wrapper library to accommodate shading.

Also, if you end up creating a shadow wrapper for OpenSceneGraph which would be relevant or helpful for DART users, we would love to get a pull request for it.

costashatz, 2018-02-06:
@dhkim0821 @mxgrey I am in the process of making a PR to enable shadows in the wrapper for OpenSceneGraph of DART. I will have it ready shortly. Here's a first glimpse of it:

[image]

mxgrey, 2018-02-06:
Thanks, @costashatz , that looks amazing!

dhkim0821, 2018-02-06:
I also tried various methods.
The followings are my current progress
[image]
[image]

I see that the control panel also becomes black because the shadow is in there too. I need to find the way to separate and apply the shadowing only for mesh.

costashatz, 2018-02-06:
@dhkim0821 great examples! Can you share the code with us? See also #976 for my try. Are you very far away from it? Thanks!

dhkim0821, 2018-02-15:
I think that the progress about shadow rendering is far from what I did. Just in case, I put the code for my shadow setting here. 

    osg::ref_ptr<osgShadow::ShadowedScene> shadowedScene = new osgShadow::ShadowedScene;
    osgShadow::ShadowSettings* settings = shadowedScene->getShadowSettings();
    settings->setReceivesShadowTraversalMask(ReceivesShadowTraversalMask);
    settings->setCastsShadowTraversalMask(CastsShadowTraversalMask);
    osg::ref_ptr<osgShadow::MinimalShadowMap> msm = new osgShadow::LightSpacePerspectiveShadowMapDB;
    shadowedScene->setShadowTechnique( msm.get() );

    float minLightMargin = 10.f;
    float maxFarPlane = 0;
    unsigned int texSize = 1024;
    unsigned int baseTexUnit = 0;
    unsigned int shadowTexUnit = 1;

    msm->setMinLightMargin( minLightMargin );
    msm->setMaxFarPlane( maxFarPlane );
    msm->setTextureSize( ::osg::Vec2s( texSize, texSize ) );
    msm->setShadowTextureCoordIndex( shadowTexUnit );
    msm->setShadowTextureUnit( shadowTexUnit );
    msm->setBaseTextureCoordIndex( baseTexUnit );
    msm->setBaseTextureUnit( baseTexUnit );

jslee02, 2018-02-17:
I believe this is addressed by #976. Thanks for the great work @costashatz and bringing up this feature request @dhkim0821.

closed 2018-02-17 by jslee02