dartsim/dart #984: Consider removing ResourceRetriever::getFilePath()

jslee02, 2018-02-12:
See comments in #972

psigen, 2018-03-03:
It seems like this is a rather long string of special cases to handle the fact that something somewhere needs a file path in order to load a resource without using our ResourceRetriever.

@jslee02 can you identify the items that are trying to do this?  Is it something with OSG?

jslee02, 2018-03-20:
Yes, it's only OSG. We need a file path to an image file that is required by OSG. [OSG roads an image from a file path](https://github.com/dartsim/dart/blob/master/dart/gui/osg/render/MeshShapeNode.cpp#L804) to use it for setting a texture of a mesh.

psigen, 2018-03-21:
Although it's a little silly, one option could be to augment the OSG
bindings with utility methods that load from string or binary by internally
pushing the content to tempfiles that are scoped to the lifetime of the
process.

I think there are some utility methods for doing this in SDL and a few
other cross platform frameworks that we could look at.

This is of course very silly, but it would let us get the API that we
actually want and we can potentially pursue a PR into OSG to add this
functionality over time as a performance optimization.