dartsim/dart #1001: Need Better Docs for Windows/VS17 Install

xemjeff, 2018-02-21:
After having recently built DART on macOSX and wrapping it in C# using Swig, I am now trying to do the same on Windows 10.  But the build/install instructions for Windows are not complete.

I'm happy to contribute a more complete doc, but I need some information first. Maybe someone wants to help.

1. What are the dependencies and how are the installed with vcpkg? We need the definitive list added to the Windows install page doc.
Here's what I have so far:  _eigen3, assimp, boost-regex, boost-system-dev, osg_
Note that boost-system and not boost-system-dev is listed on the vcpkg ports lists.

2. The Windows installation page on DART says to use VS 2017 at the top and then later says to have Visual Studio 14 installed (that's VS 2015). That page needs updating.
_Click “Configure” and choose your development environment (Visual Studio 14)._

3. It would be very helpful to have a set of x64 binaries and libraries all built that would let people quickly try out and experiment with DART.  PyDart/PyDart2 included would make quick use even easier.

If we want to see DART used by more people, then providing an easy way to build or install x64 binaries on Windows will definitely grow the user base.

Jeff.

jslee02, 2018-02-21:
Thanks for the feedback. Here are the answers to your questions.

1. The [dependencies](https://github.com/dartsim/dart/blob/master/.appveyor.yml#L40-L58) are as follows:
    * Required: boost, eigen3, ccd, fcl, assimp
    * Optional: nlopt, ipopt, bullet3, ode, flann, tinyxml2, urdfdom_headers, urdfdom, opengl, freeglut, osg ([one of the osg dependency had an issue](https://github.com/Microsoft/vcpkg/issues/2551), but I haven't tested it recently)

1. Inconsistent VS versions will be updated.

1. That could be a good option, but I'm not sure if we have enough time to maintain binaries manually. We used to provide sets of binaries of the DART dependencies, but we were required to provide the binaries for different combinations of VS versions, VS options (e.g., /MT, /MD), x86/x64 etc, which were too much burdensome for limited manpower here. An ideal way of doing that would be using a package manager like apt-get and Homebrew, but there was no reliable package manager on Windows. Now we have [vcpkg with all the DART dependencies](https://github.com/dartsim/dart/issues/260) except [ipopt](https://github.com/Microsoft/vcpkg/issues/233). So we can try to submit DART port to vcpkg. I'm not sure if vcpkg support python packages as well, though.

Wrapping DART in C# sounds awesome. It would be grateful if you could share it once it's ready. :smile:

xemjeff, 2018-02-21:
Thanks for the fast reply!

*Boost:*
It's quit large to pull in everything.
Is DART just using boost-regex, boost-system from boost?

*C# Wrapper:*
I'll create an open github repository for the C# wrapper (similar to
PyDart) when it's ready.
My plan is to integrate with game engines Unity and Godot, so I don't plan
to wrap the GUI part of DART.

Jeff.

jslee02, 2018-02-21:
> It's quit large to pull in everything.
> Is DART just using boost-regex, boost-system from boost?

Oh, I didn't know that vcpkg has separated boost ports. Yeah, DART only depends on boost-regex, boost-system, and boost-filesystem. Thanks!

> I'll create an open github repository for the C# wrapper (similar to
> PyDart) when it's ready.
> My plan is to integrate with game engines Unity and Godot, so I don't plan
> to wrap the GUI part of DART.

All sounds great!

mxgrey, 2018-02-21:
I'm pretty sure `boost-regex` is only needed for certain versions of GCC, and should not necessary on Windows, because for that we can just use `std::regex`.

jslee02, 2018-02-21:
@mxgrey That's right. I guess we have to modify [these lines](https://github.com/dartsim/dart/blob/d7afcacdcdcf7da688fb61caf1761b1309888ffe/cmake/DARTFindDependencies.cmake#L91-L95) so that CMake searches `boost-regex` only when it's not `MSVC`. Will apply this to https://github.com/dartsim/dart/pull/1002.

jslee02, 2018-08-24:
Closing this issue. Feel free to reopen if you have further concerns.

closed 2018-08-24 by jslee02