dartsim/dart #1005: Submit DART port to vcpkg

jslee02, 2018-02-22:
[As most of the dependencies of DART are added to vcpkg](https://github.com/dartsim/dart/issues/260), it would be good to submit DART port to [vcpkg](https://github.com/Microsoft/vcpkg) as well so that the users on Windows can easily install DART using the package manager.

smartcam3d, 2018-02-25:
I am trying to follow the instructions for the installation and the tutorials under windows MSVC 2017. I managed to install all the dependencies with vcpkg and to build the library. Unfortunatelly I did not manage to run the examples yet. After setting the compiler flag -permissive-, the tutorial code for the pendulum is compiled, but crashes on execution. I would appreciate more detailed instructions on how to use DART on windows and think that a vcpkg integration would be very helpful!

Fleischner, 2018-03-22:
Currently trying to do that. 
One problem is building as shared library fails. 

Dart does not mark its exports with __declspec(dllexport) as windows requires. I use the cmake option WINDOWS_EXPORT_ALL_SYMBOLS to generate an .def to work around that. However in this case the dart.dll has more than 65535 symbols. And this is not supported by msvc compiler (LNK1189) https://msdn.microsoft.com/en-us/library/aa234479(v=vs.60).aspx

Most of the symbols result from eigen3 header only library. Maybe upcoming solution I proposed at cmake: https://gitlab.kitware.com/cmake/cmake/issues/17841

jslee02, 2018-03-22:
@Fleischner Thanks for working on this.

A straightforward solution would be that DART marks `__declspec(dllexport)` using CMake's [GenerateExportHeader](https://cmake.org/cmake/help/v3.0/module/GenerateExportHeader.html) like [FCL did](https://github.com/flexible-collision-library/fcl/pull/233/files). I can do this if this sounds good.

mxgrey, 2018-03-22:
I've been meaning to recommend that DART use `GenerateExportHeader`, so +1 to that.

jslee02, 2018-03-22:
@smartcam3d Let me try to reproduce the problem when I have a chance. Issue #1039 created to track the problem.

Fleischner, 2018-03-22:
I didn't know about GenerateExportHeader. This is of course a better solution.

mxgrey, 2018-03-22:
In the meantime, it should still be possible to use DART as a static library. Is vcpkg able to support libraries that can only be built as static libraries?

Fleischner, 2018-03-22:
Yes it is. It can even mix. 
You can look at the triplets https://github.com/Microsoft/vcpkg/blob/master/docs/users/triplets.md
or https://github.com/Microsoft/vcpkg/tree/master/triplets

When vcpkg calls darts cmake file it passes in BUILD_SHARED_LIBS variable. You can build static on shared by overriden the variable.

As default there is shared and static builds. For DART this would be e.g:
vcpkg install dartsim:x64-windows -> need to override BUILD_SHARED_LIBS until exports fixed
vcpkg install dartsim:x64-windows-static

jslee02, 2018-03-22:
@Fleischner #1040 is created to explicitly specify which API to be exposed as discussed above.

Anyone knows how to count the number of symbols of a shared library? I would like to see the difference so that I'm sure #1040 is worthwhile. :smile:

Fleischner, 2018-03-22:
@jslee02 . Wow that was fast 👍

For counting there is sure an clean way to do this. I found at least an 'ugly' one yesterday:

build with cmake and set -DCMAKE_WINDOWS_EXPORT_ALL_SYMBOLS. You will get an .def file with as many lines as symbols. If I remeber correctly dart had around 90k.

Fleischner, 2018-03-31:
Building with #1040 

G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/common/AspectWithVersion.hpp(55): error C2668: 'dart::common::detail::AspectWithVersionedProperties<dart::common::CompositeTrackingAspect<CompositeT>,DerivedT,PropertiesDataT,CompositeT,void dart::common::detail::NoOp<DerivedT*>(DerivedT *)>::AspectWithVersionedProperties': ambiguous call to overloaded function
        with
        [
            CompositeT=dart::dynamics::ShapeFrame,
            DerivedT=dart::dynamics::VisualAspect,
            PropertiesDataT=dart::dynamics::detail::VisualAspectProperties
        ]
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/common/detail/AspectWithVersion.hpp(124): note: could be 'dart::common::detail::AspectWithVersionedProperties<dart::common::CompositeTrackingAspect<CompositeT>,DerivedT,PropertiesDataT,CompositeT,void dart::common::detail::NoOp<DerivedT*>(DerivedT *)>::AspectWithVersionedProperties(const dart::dynamics::detail::VisualAspectProperties &)'
        with
        [
            CompositeT=dart::dynamics::ShapeFrame,
            DerivedT=dart::dynamics::VisualAspect,
            PropertiesDataT=dart::dynamics::detail::VisualAspectProperties
        ]
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/common/detail/AspectWithVersion.hpp(120): note: or       'dart::common::detail::AspectWithVersionedProperties<dart::common::CompositeTrackingAspect<CompositeT>,DerivedT,PropertiesDataT,CompositeT,void dart::common::detail::NoOp<DerivedT*>(DerivedT *)>::AspectWithVersionedProperties(void)'
        with
        [
            CompositeT=dart::dynamics::ShapeFrame,
            DerivedT=dart::dynamics::VisualAspect,
            PropertiesDataT=dart::dynamics::detail::VisualAspectProperties
        ]
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/dynamics/ShapeFrame.hpp(111): note: while trying to match the argument list '()'
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/dynamics/ShapeFrame.hpp(111): note: This diagnostic occurred in the compiler generated function 'void dart::common::detail::AspectWithVersionedProperties<dart::common::CompositeTrackingAspect<CompositeT>,DerivedT,PropertiesDataT,CompositeT,void dart::common::detail::NoOp<DerivedT*>(DerivedT *)>::__dflt_ctor_closure(void)'
        with
        [
            CompositeT=dart::dynamics::ShapeFrame,
            DerivedT=dart::dynamics::VisualAspect,
            PropertiesDataT=dart::dynamics::detail::VisualAspectProperties
        ]
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/dynamics/ShapeFrame.hpp(119): warning C4251: 'dart::common::detail::AspectWithVersionedProperties<dart::common::CompositeTrackingAspect<CompositeT>,DerivedT,PropertiesDataT,CompositeT,void dart::common::detail::NoOp<DerivedT*>(DerivedT *)>::mProperties': class 'dart::common::MakeCloneable<dart::common::Aspect::Properties,dart::dynamics::detail::CollisionAspectProperties>' needs to have dll-interface to be used by clients of class 'dart::common::detail::AspectWithVersionedProperties<dart::common::CompositeTrackingAspect<CompositeT>,DerivedT,PropertiesDataT,CompositeT,void dart::common::detail::NoOp<DerivedT*>(DerivedT *)>'
        with
        [
            CompositeT=dart::dynamics::ShapeFrame,
            DerivedT=dart::dynamics::CollisionAspect,
            PropertiesDataT=dart::dynamics::detail::CollisionAspectProperties
        ]
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/common/Cloneable.hpp(84): note: see declaration of 'dart::common::MakeCloneable<dart::common::Aspect::Properties,dart::dynamics::detail::CollisionAspectProperties>'
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/dynamics/ShapeFrame.hpp(140): warning C4251: 'dart::common::detail::AspectWithVersionedProperties<dart::common::CompositeTrackingAspect<CompositeT>,DerivedT,PropertiesDataT,CompositeT,void dart::common::detail::NoOp<DerivedT*>(DerivedT *)>::mProperties': class 'dart::common::MakeCloneable<dart::common::Aspect::Properties,dart::dynamics::detail::DynamicsAspectProperties>' needs to have dll-interface to be used by clients of class 'dart::common::detail::AspectWithVersionedProperties<dart::common::CompositeTrackingAspect<CompositeT>,DerivedT,PropertiesDataT,CompositeT,void dart::common::detail::NoOp<DerivedT*>(DerivedT *)>'
        with
        [
            CompositeT=dart::dynamics::ShapeFrame,
            DerivedT=dart::dynamics::DynamicsAspect,
            PropertiesDataT=dart::dynamics::detail::DynamicsAspectProperties
        ]
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/common/Cloneable.hpp(84): note: see declaration of 'dart::common::MakeCloneable<dart::common::Aspect::Properties,dart::dynamics::detail::DynamicsAspectProperties>'
G:\code\vcpkg\buildtrees\dart\src\dart-build-visibility\dart/dynamics/ShapeFrame.hpp(162): warning C4251: 'dart::common::SpecializedForAspect<ReqAspect>::mSpecAspectIterator': class 'std::_Tree_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>>' needs to have dll-interface to be used by clients of class 'dart::common::SpecializedForAspect<ReqAspect>'
        with
        [
            ReqAspect=dart::common::EmbeddedPropertiesAspect<dart::dynamics::ShapeFrame,dart::dynamics::detail::ShapeFrameProperties>
        ]
        and
        [
            _Ty=std::pair<const std::type_index,std::unique_ptr<dart::common::Aspect,std::default_delete<dart::common::Aspect>>>
        ]
        and
        [
[... cut ...]
        with
        [
            _Ty=std::pair<const std::type_index,std::unique_ptr<dart::common::Aspect,std::default_delete<dart::common::Aspect>>>
        ]
e:\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xtree(561): note: see declaration of 'std::_Tree_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>>'
        with
        [
            _Ty=std::pair<const std::type_index,std::unique_ptr<dart::common::Aspect,std::default_delete<dart::common::Aspect>>>
        ]

Fleischner, 2018-03-31:
And there are countless warnings C4251: needs to have dll-interface to be used by clients of class. So best to disable this warning.

Most of these C4251 derive from std:: and eigen3:: (if these types are members of classes that are exported, even if members are private). This is a general problem with c++. 

But as far as I understand C4251 is no problem here. Std:: should be fine as long as respecting the general requirements of building everything with the exact matching compiler version against the same standard library. And eigen3 should be fine as it is header only.

The question is how to easily find the C4251 that are triggered by missing exports in dart? There is a new filter option in visual studio to silence warnings be external headers. Maybe this helps. I need to try this.

traversaro, 2020-09-02:
> 
> 
> In the meantime, it should still be possible to use DART as a static library. Is vcpkg able to support libraries that can only be built as static libraries?

Yes, this is possible by starting the port with ` vcpkg_check_linkage(ONLY_STATIC_LIBRARY)`.  Related PR: https://github.com/microsoft/vcpkg/pull/13320 .

traversaro, 2020-09-03:
Some random issues I found blocking/nice to have for the vcpkg port: 
* [ ] (Blocking): Modify dart_check_optional_package  ( https://github.com/dartsim/dart/blob/e480e2f06bd8c0dbd0be6862b7e5b4cb01df6cdf/cmake/DARTMacros.cmake#L129 ) to observe a variable `DART_SKIP_<dep>` that, if set to true force not to use a dependency even if it is found in the system. This is required for reproducibility, and ensure that the same version of DART is compiled in vcpkg regardless of the installed ports.
* [ ] (Nice to have): Bump octomap version in vcpkg to 1.9.1 to remove workadound https://github.com/dartsim/dart/blob/becbeada2a8ba94834e69ceb9f4efaa92381b756/cmake/DARTFindDependencies.cmake#L93

traversaro, 2020-09-03:
>     * (Blocking): Modify dart_check_optional_package  ( https://github.com/dartsim/dart/blob/e480e2f06bd8c0dbd0be6862b7e5b4cb01df6cdf/cmake/DARTMacros.cmake#L129

Done in https://github.com/dartsim/dart/pull/1497 .

traversaro, 2020-09-04:
>     * (Nice to have): Bump octomap version in vcpkg to 1.9.1 to remove the workaround in https://github.com/dartsim/dart/blob/becbeada2a8ba94834e69ceb9f4efaa92381b756/cmake/DARTFindDependencies.cmake#L93

Done in https://github.com/microsoft/vcpkg/pull/13356 .

traversaro, 2020-09-15:
As https://github.com/microsoft/vcpkg/pull/13320 has been merged, `dartsim` is now an officially supported port of vcpkg.

jslee02, 2020-09-15:
Awesome! Thank you for your effort on this matter, @traversaro! Let me review https://github.com/dartsim/dart/pull/1497 soon.

closed 2020-09-15 by jslee02