dartsim/dart #1007: Adding a mutex to MetaSkeleton

brianhou, 2018-02-22:
In https://github.com/personalrobotics/aikido, we're currently opting to use `MetaSkeleton`s rather than `Skeleton`s. However, `MetaSkeleton`s don't have associated mutex(es) so I'm concerned we'll run into issues with thread safety.

It seemed like there was some discussion in #466 about what to do with `ReferentialSkeleton`s on that PR, but I don't think a `getMutexes()` method was ever added.

mxgrey, 2018-02-22:
I edited the title for clarity. It wouldn't be helpful to add a mutex to `MetaSkeleton`, but a function like `getMutexes()` would be useful. The hard part is making sure that we have a good locking scheme for dealing with a set of mutexes instead of just a single mutex.

jslee02, 2018-02-26:
@brianhou, @mxgrey, @mkoval, @gilwoolee I've create a WIP PR #1011 for this. Could you take a look at it when you guys have a chance? Thanks!

jslee02, 2018-05-12:
Resolved by #1011

PR #1011 (2018-05-01): Add LockableReference classes
This PR is a minimal set of changes of adding `MetaSkeleton::getMutexes()` with two helper functions to lock and unlock mutexes.

* Related PR: #466 
* Resolves #1007 

**Possible Improvement**
As pointed out by this [article](https://www.justsoftwaresolutions.co.uk/threading/acquiring-multiple-locks-without-deadlock.html), locking by address works for small numbers of mutexes, but it could be inefficient for many mutexes. Suggested improvement is using "try-and-back-off". But this PR leaves this a future improvement since it could be achieved without the public A [... cut]
closed 2018-05-12 by jslee02