dartsim/dart #1008: API questions on updating skeleton (computeForwardKinematics())

junhyeokahn, 2018-02-22:
Hello,
I have small questions on how dart updates skeleton.
I would like to update forward kinematics only at the specific time based on my sensor data.
So what I've tried was once I got sensor data (q and qdot), I set positions and velocities using setPositions and setVelocities functions. After that, I am trying to compute centroid dynamics related properties of skeleton. However, in the process, it takes longer time than I expected and I think there is some redundant kinematics updating underneath.

According to the explanation on computeForwardKinematics  in API document, dart seems only compute forward kinematics once if I call ```computeForwardKinematics()``` method directly after I set position and velocity. However, I feel like dart keeps updating forward kinematics whenever I require kinematics related functions (e.g. getCOM() ), so some amount of time is spent.

Please let me know if I misunderstand some part.
Thank you.

mxgrey, 2018-02-22:
To clarify, are you calling `Skeleton::getCOM()` or `BodyNode::getCOM()`?

The function `Skeleton::getCOM()` always has O(N) complexity in the number of BodyNodes, because we do not cache the Skeleton's center of mass (there would be a lot of variables to track for us to do that, and I don't think it would be worth the bookkeeping cost).

The function `BodyNode::getCOM()` should only involve a multiplying a vector by a transformation matrix if the kinematics have already been updated. If you believe this function is always triggering a kinematics update, despite the kinematics already being updated, then that would be concerning.

We could probably do a better job of documenting the expected computational complexity of each function.

If there are specific functions that you think are not doing their lazy evaluation correctly, a list would be appreciated. If you think that none of the lazy evaluation is working correctly, then that would be pretty alarming, and we should try to come up with some performance tests to investigate that.

mxgrey, 2018-02-22:
Also, just to make sure we're ruling out this possibility: If you're using DART after compiling it from source, be sure to build it in `Release` or `RelWithDebInfo` mode. I think the default build mode is `Debug`, which will always run rather slowly because Eigen does not have very good performance in debug mode.

Just add `-DCMAKE_BUILD_TYPE=Release` or `-DCMAKE_BUILD_TYPE=RelWithDebInfo` when you call `cmake` from the command line.

jslee02, 2018-02-23:
@mxgrey Good point! But the default is actually `Release`. [We use `Release` mode when `CMAKE_BUILD_TYPE` is not provided](https://github.com/dartsim/dart/blob/master/CMakeLists.txt#L111-L113).

mxgrey, 2018-02-23:
Ah, okay, I'm thinking of the native cmake default rather than DART's default.

However, if you're creating an application that uses DART and cmake, then you should make sure to compile your own application in `Release` mode, because you'll still experience some slowdowns from compiling Eigen in `Debug` mode. And unless you do something to explicitly make `Release` the default build mode, cmake will have your default be `Debug`.

junhyeokahn, 2018-02-23:
@mxgrey Thanks to your comment, ```-DCMAKE_BUILD_TYPE=Release``` resolves the problem. I guess I installed dart6.4 in my system and use it as a package and I did not specify build type for my project before. Thanks :)

closed 2018-02-23 by junhyeokahn