dartsim/dart #1023: Questions on APIs related to collision shape

junhyeokahn, 2018-03-04:
Hi,
I have small question on APIs related to collision shape.
If I want to get a size of collision shape (e.g. height, width, length of box) of specific BodyNode, which API should I use?
I've looked into ShapeNode and Collision namespace on the document but could find one.

jslee02, 2018-03-04:
`ShapeNode::getShape()` returns the shape pointer, and you can get the properties from it. Assuming the shpe is `BoxShape`, you could do:
```c++
auso shape = shapeNode->getShape();
auto box = std::static_pointer_cast<BoxShape>();
box->getSize();
```

This means that you should know the shape type or try down-casting using `std::dynamic_pointer_cast<T>` to figure out the shape type, which I believe it's a relatively rare case.

junhyeokahn, 2018-03-04:
In the case of using a Mesh for visualization but using a Box shape for collision and if I want to get a shape of collision box, I guess I should use ```CollisionGroup::getShapeFrame(std::size_t index)``` then use ```ShapeFrame::getShape()```. But CollisionGroup may include many collision objects in the case of humanoid and how could I specify a collision link by index?

jslee02, 2018-03-04:
I'm not sure if I get your question correctly. What do you mean by "specify a collision link by index"? Basically, `CollisionGroup` is a container that you may want to perform collision checking, but it doesn't include the collision checking results. You would like to use `CollisionResult` to get the result instead.

1. If you want to get the list of `BodyNode`s in contact, you could use `CollisionResult::getCollidingBodyNodes`.

1. If you want to know more detailed results, you could use `CollisionResult::getContact(index)` that returns `Contact`. `Contact` includes the contact point, contact normal, penetration depth, and the pair of colliding `CollisionObject`s. You can then get the `ShapeFrame` using `CollisionObject::getShapeFrame()`.

Finally, if you want to know the `BodyNode` associated with the `ShapeFrame`, you could do this:
```c++
auto shapeNode = collisionObject->getShapeFrame()->asShapeNode();
if (shapeNode)
  auto bodyNode = shapeNode->getBodyNode();
```
This can be seen a bit verbose, but this is because to support collision detection not only for `BodyNode` but also any object has `ShapeFrame`.

jslee02, 2018-03-04:
I think now I get what you mean.

DART has two ways of collision checking with collision group: (1) colliding shape frames within a `CollisionGroup` and (2) colliding shape frames from two different `CollisionGroup`. We don't provide the index to the `ShapeFrame`s in a `CollisionGroup` because of (2). The index would be ambiguous in (2) case. Instead, we provide a list of `Contact` that you can get from `CollisionResutl`.

mxgrey, 2018-03-04:
> For example, I would like to know the size collision box attached to the 'rightFoot' BodyNode in the skeleton

You can get a list of the ShapeNodes that are used in collisions from the BodyNode API:

```
std::vector<dart::dynamics::ShapeNode*> shapeNodes = robot->getBodyNode("body node name")->getShapeNodesWith<dart::dynamics::CollisionAspect>();
```

As JS said, you can get the `Shape` from the `ShapeNode` using `ShapeNode::getShape()`, and then try to dynamically cast each one to a Box to find which collision shape in the body node is a box.

Or if you know the index of the ShapeNode within the BodyNode, you can do:

```
dart::dynamics::ShapeNode* sn = robot->getBodyNode("name of body node")->getShapeNode(shapeNodeIndex);
```

I expect the ShapeNode index will be based on the order in which the visual/collision shape is listed in the model format (.urdf, .sdf, or .skel), starting from the index 0. Note that visual shapes and collision shapes are indexed together (e.g. if you have 2 visual shapes and 2 collision shapes, your ShapeNode indices will go from 0 to 3), but when you do `BodyNode::getShapeNodesWith<CollisionAspect>()`, that `std::vector` of ShapeNodes will be indexed purely based on the collision shapes in the model.

mxgrey, 2018-03-04:
Also, if you know you only have exactly one collision shape in the BodyNode, you could safely do:

```
dart::dynamics::ShapeNode* sn = robot->getBodyNode("rightFoot")->getShapesWith<dart::dynamics::CollisionAspect>().front();
```

junhyeokahn, 2018-03-04:
@jslee02 @mxgrey  Thanks for your kind reply!

closed 2018-03-04 by junhyeokahn