dartsim/dart #1035: Eigen version 3.2.1-3.28 compilation issue

costashatz, 2018-03-21:
Hello,

There are a few issues with the file `dart/common/Memory.hpp`. First of all, in the master branch the includes are being included twice: see [here](https://github.com/dartsim/dart/blob/master/dart/common/Memory.hpp#L40).

Secondly, I am getting the following issues for these lines of code
https://github.com/dartsim/dart/blob/4b38f46f3009066ce8c8e605964842354b02e308/dart/common/Memory.hpp#L51-L61

```bash
[  3%] Built target dart-external-odelcpsolver

[  4%] [  4%] Built target dart-external-lodepng

Built target dart-external-imgui

[  4%] [  5%] [  5%] Building CXX object dart/CMakeFiles/dart.dir/common/Composite.cpp.o

Building CXX object dart/CMakeFiles/dart.dir/common/Subject.cpp.o

[  6%] Building CXX object dart/CMakeFiles/dart.dir/common/LocalResourceRetriever.cpp.o

Building CXX object dart/CMakeFiles/dart.dir/common/Timer.cpp.o

[  6%] Building CXX object dart/CMakeFiles/dart.dir/common/SharedLibrary.cpp.o

[  6%] Building CXX object dart/CMakeFiles/dart.dir/common/VersionCounter.cpp.o

[  8%] Building CXX object dart/CMakeFiles/dart.dir/common/Observer.cpp.o

[  8%] Building CXX object dart/CMakeFiles/dart.dir/common/ResourceRetriever.cpp.o

[  8%] Building CXX object dart/CMakeFiles/dart.dir/common/Console.cpp.o

In file included from /home/supaul/dart/dart/common/Cloneable.hpp:39:0,

                 from /home/supaul/dart/dart/common/Aspect.hpp:38,

                 from /home/supaul/dart/dart/common/detail/CompositeData.hpp:43,

                 from /home/supaul/dart/dart/common/Composite.hpp:36,

                 from /home/supaul/dart/dart/common/Composite.cpp:37:

/home/supaul/dart/dart/common/Memory.hpp:55:5: error: ‘detail’ is not a member of ‘dart::common’

     dart::common::detail::aligned_allocator_cpp11<_Tp>>;

     ^

/home/supaul/dart/dart/common/Memory.hpp:55:5: error: ‘detail’ is not a member of ‘dart::common’

/home/supaul/dart/dart/common/Memory.hpp:55:54: error: template argument 2 is invalid

     dart::common::detail::aligned_allocator_cpp11<_Tp>>;

                                                      ^

/home/supaul/dart/dart/common/Memory.hpp:59:5: error: ‘detail’ is not a member of ‘dart::common’

     dart::common::detail::aligned_allocator_cpp11<std::pair<const _Key, _Tp>>>;

     ^

/home/supaul/dart/dart/common/Memory.hpp:59:5: error: ‘detail’ is not a member of ‘dart::common’

/home/supaul/dart/dart/common/Memory.hpp:59:78: error: template argument 4 is invalid

     dart::common::detail::aligned_allocator_cpp11<std::pair<const _Key, _Tp>>>;

                                                                              ^

[  9%] Building CXX object dart/CMakeFiles/dart.dir/common/Aspect.cpp.o

[  9%] Building CXX object dart/CMakeFiles/dart.dir/common/detail/ConnectionBody.cpp.o

[  9%] Building CXX object dart/CMakeFiles/dart.dir/common/detail/SharedLibraryManager.cpp.o

make[2]: *** [dart/CMakeFiles/dart.dir/common/Composite.cpp.o] Error 1

make[2]: *** Waiting for unfinished jobs....

In file included from /home/supaul/dart/dart/common/Cloneable.hpp:39:0,

                 from /home/supaul/dart/dart/common/Aspect.hpp:38,

                 from /home/supaul/dart/dart/common/Aspect.cpp:37:

/home/supaul/dart/dart/common/Memory.hpp:55:5: error: ‘detail’ is not a member of ‘dart::common’

     dart::common::detail::aligned_allocator_cpp11<_Tp>>;

     ^

/home/supaul/dart/dart/common/Memory.hpp:55:5: error: ‘detail’ is not a member of ‘dart::common’

/home/supaul/dart/dart/common/Memory.hpp:55:54: error: template argument 2 is invalid

     dart::common::detail::aligned_allocator_cpp11<_Tp>>;

                                                      ^

/home/supaul/dart/dart/common/Memory.hpp:59:5: error: ‘detail’ is not a member of ‘dart::common’

     dart::common::detail::aligned_allocator_cpp11<std::pair<const _Key, _Tp>>>;

     ^

/home/supaul/dart/dart/common/Memory.hpp:59:5: error: ‘detail’ is not a member of ‘dart::common’

/home/supaul/dart/dart/common/Memory.hpp:59:78: error: template argument 4 is invalid

     dart::common::detail::aligned_allocator_cpp11<std::pair<const _Key, _Tp>>>;

                                                                              ^

make[2]: *** [dart/CMakeFiles/dart.dir/common/Aspect.cpp.o] Error 1

make[1]: *** [dart/CMakeFiles/dart.dir/all] Error 2

make: *** [all] Error 2
```

It somewhat makes sense, because the `detail::common` namespace is not being included in this file (of course I did not trace all the possible chains).

This happens both in `master` and `release-6.4` branches.. I guess you need to include this file `"dart/common/detail/AlignedAllocator.hpp"`?

Just reporting the issue..

mxgrey, 2018-03-21:
Thanks for the report!

Is this issue occurring when compiling DART, or only when compiling your own codebase against DART?

If it's the latter, does it happen when you're manually including the `dart/common/Memory.hpp` header in one of your source files, or is it happening when including a different DART header that uses `Memory.hpp`?

costashatz, 2018-03-21:
> Is this issue occurring when compiling DART, or only when compiling your own codebase against DART?

When compiling dart.. I didn't (couldn't really) try to compile my code against dart..

jslee02, 2018-03-25:
This should be resolved by #1042 and https://github.com/dartsim/dart/commit/e2b686ad5998309a8b1dc27a76ebbbf6a2f4440e. Thanks for the report again!

jslee02, 2018-03-28:
Closing. Please reopen if this issue persists.

closed 2018-03-28 by jslee02

costashatz, 2018-03-28:
@jslee02 thanks for the ultra fast fix!

PR #1042 (2018-03-25): Fix build with Eigen 3.2.1-3.2.8
This PR partially resolves #1035.

***

**Before creating a pull request**

- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`