dartsim/dart #1037: Add educational tutorials for fundamental control theory and numeric methods

mxgrey, 2018-03-21:
DART is a very useful toolkit for implementing sophisticated controllers and easily running physics simulations, but a basic familiarity with control theory and numerical methods is still necessary in order to get sane results. Since the robotics community consists of people from a broad variety of backgrounds, it would be useful for DART to provide some introductory educational tutorials to familiarize beginners to these important concepts. This would be especially valuable in reducing the barrier of entry into the robotics research world.

Since DART's API tries to rigorously parallel the conventional terminology in mechanical engineering and control theory, DART may be especially well suited as an educational foundation for students and beginners.

vardhinipawar, 2020-05-22:
Can I work on this?

mxgrey, 2020-05-26:
That would be wonderful, @vardhinipawar! Any contributions towards this effort would be broadly appreciated by the community :+1: