dartsim/dart #1045: Build Failing

G-Pereira, 2018-03-30:
#### Environment
DART 6.4.0
OS Arch Linux
GCC 7.3.1

#### Expected Behavior
Compile successfully

#### Current Behavior
https://gist.github.com/G-Pereira/11c42d4fb2273059560787a66f318ab2

#### Steps to Reproduce
Compile it

jslee02, 2018-03-30:
Thanks for the report.

I have no experience with Arch Linux. Is the first line (`dart-6.4.0 make -j8 VERBOSE=1`) a command you used to build DART? Did you configure using CMake before run `make`?

It would be helpful if you post what's the output of following procedure:
```shell
$ cd <dart_dir>
$ mkdir build && cd build # we recommend out-of-source build
$ cmake ..
$ make -j8
```

Also, it seems you configured DART in-source, but we recommend building DART out-of-source (by making `build` directory for example).

G-Pereira, 2018-04-01:
Thanks for the reply!
I will try this in the next 24 hours, and will report back!

jslee02, 2018-04-01:
@G-Pereira I don't know why, but it seems `dynamics.hpp` includes itself, which shouldn't do. Could you post the output of `$ cat /home/gpereira/Downloads/dart-6.4.0/dart/dynamics/dynamics.hpp` to ensure this?

mxgrey, 2018-04-03:
Try completely deleting your DART source code and starting over, this time using an out-of-source build directory.

If you build in-source, then you'll end up with a `dart/dynamics/dynamics.hpp` file in your source folder. Once that happens, every time you attempt to configure, it will add that header to the list of headers that are supposed to go into `dart/dynamics/dynamics.hpp`. Starting over with an out-of-source build won't help if you've already corrupted the source directory.

mxgrey, 2018-04-06:
I'm going to go ahead and close this issue. Feel free to reopen it if starting from scratch and using an out-of-source build does not fix the issue.

G-Pereira, 2018-04-25:
it continues throwing the same error @mxgrey 
And that file does not exist @jslee02 only the one showed in the picture. Even before compiling.
[image]

G-Pereira, 2018-04-25:
[image]

mxgrey, 2018-04-25:
Are you building with multiple jobs? From what I can tell, there seem to be two issues, which may be independent of each other:

1. A linking error related to `flann`.
2. A header include recursion error related to `dynamics.hpp`.

The latest screenshot shows problem (1), but it does not seem to show problem (2). The out-of-source build should have fixed problem (2), so I'd like to clarify whether that particular problem is still an issue.

G-Pereira, 2018-04-25:
Let me compile with one job, I was compiling with 8 jobs

G-Pereira, 2018-04-25:
The issue remains the same

G-Pereira, 2018-04-25:
@mxgrey Any clues?

mxgrey, 2018-04-25:
Do you mean you're still getting the issue where it recursively includes `dynamics.hpp`? Because your latest screenshot does not appear to have that issue, and I believe it should be fixed by doing an out-of-source build.

It seems that the `flann` linking issue is unrelated, and is a [known issue in Arch Linux](https://aur.archlinux.org/packages/flann/).

We don't officially support Arch Linux, but if the fix would be simple enough, I think we'd be happy to accept it. It seems like we need to add something along the lines of `target_link_libraries(${target_name} lz4)` to the [dart/planning/CMakeLists.txt file](https://github.com/dartsim/dart/blob/release-6.3/dart/planning/CMakeLists.txt#L24). Give that a try and let us know if it solves the problem for you.

Alternatively, it looks like the current master (which is the testing ground for release-7.0) has removed the `dart-planning` component entirely, so you could try building the current `master` branch and see if you run into any issues.

jslee02, 2018-04-25:
> Alternatively, it looks like the current master branch of DART (which is the testing ground for release-7.0) has removed the dart-planning component entirely, so you could try building the current master branch and see if you run into any issues.

Also, if you don't use `flann` besides DART, you could uninstall it so that DART just don't build the planning component.

jslee02, 2018-04-25:
> It seems like we need to add something along the lines of target_link_libraries(${target_name} lz4) to the dart/planning/CMakeLists.txt file. Give that a try and let us know if it solves the problem for you.

Also, we could modify `FindFLANN.cmake` to add `lz4` into `FLANN_LIBRARIES`.

G-Pereira, 2018-04-26:
Hi! I am now maintainer of flann package in Arch Linux, I can fix this on the flann side ;)

jslee02, 2019-04-29:
Any luck on flann package? Closing for now because [`libdart` package seems to work](https://aur.archlinux.org/packages/libdart/). Feel free to reopen if there is a further issue.

closed 2019-04-29 by jslee02