dartsim/dart #1047: Consider supporting transmission constraints

mxgrey, 2018-03-30:
Real robots often have transmissions between joints, which may have properties like gearing ratios and backlash. There can also be mechanisms where multiple joints are coupled. A common example of this is underactuated robot grippers where a 1-DOF motor controls several multi-jointed fingers simultaneously.

I don't know how difficult it would be to implement this type of constraint on top of Featherstone's ABA, but I think it would be good for us to investigate it, and implement it if it turns out to be feasible.

acxz, 2019-02-17:
Hello, I am working on a robot where we would like to be able to model transmissions. I was wondering what the current progress on this topic is and if it is being pursued. We know that we can add a transmission property in our urdf, but it would be nice if DART can model this property as well.

jslee02, 2019-02-18:
This sounds like mimic joints ([defined by URDF](http://wiki.ros.org/urdf/XML/joint)), which is already implemented by @costashatz in #1178! @mxgrey Could you verify if you meant it and close this issue if so?

acxz, 2019-02-18:
Here is the [urdf page on transmissions](http://wiki.ros.org/urdf/XML/Transmission) to clarify on what I meant by transmissions.

mxgrey, 2019-02-18:
I believe the mimic joint added in #1178 can satisfy what I meant by "transmission constraint" in this issue.

The URDF concept of a "transmission" is a bit different, because it seems the joint force is being transmitted from an "actuator". DART doesn't currently make a significant distinction between an actuator and a joint. Also the mimic joint is a position-based constraint whereas the URDF transmission seems to be a power-based constraint.

costashatz, 2019-02-18:
We can easily integrate the [SimpleTransmission](https://github.com/ros-controls/ros_control/blob/melodic-devel/transmission_interface/include/transmission_interface/simple_transmission.h) interface that ROS has (and is the most frequent). But then for each different transmission, we should create different constraints (as they do in ROS). If you want, I can do it for the simple transmission and see how it works..

junhyeokahn, 2021-10-18:
@costashatz @mxgrey 
Hello, just have a quick question on the mimic joints.
It looks like mimic joints enforce a position constraint, but does it also transmit the force?
In other words, if I apply 1N to one joint, is the same force applied to the other joint with the position (and velocity) mimicking satisfied?

closed 2025-11-26 by jslee02