dartsim/dart #1052: [Question] URDF collision objects

EastskyKang, 2018-04-10:
I am using ```DartLoader``` to load my URDF file. (quadruped robot)

I want to use primitive collision shapes for each link. For instance, base link of my robot is defined as follow:

```xml
<link name="base">
        <visual>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="anymal_base_2.dae" scale="0.001 0.001     0.001"/>
            </geometry>
            <material name="anymal_material"/>
        </visual>
        <collision>
            <!-- Main Body -->
            <origin rpy="0 0 0" xyz="0 0 0.08"/>
            <geometry>
                <box size="0.531 0.232 0.26"/>
            </geometry>
        </collision>
    </link>
```

However, as I load URDF file by ```DartLoader```, it seems the root bodyNode (which corresponds to the base link) has two shapes, mesh and box which is not expected. (Since it has only one collision node, bodyNode should have only one shape)

Is there anyway to create collision objects which only correspond to ```<collision>``` node?

mxgrey, 2018-04-10:
I assume you're using DART 6 or higher, please let us know if that's not the case.

This URDF will result in the BodyNode named "base" having two ShapeNodes attached to it

1. One of the ShapeNodes attached to "base" will correspond to the visual geometry defined by `<visual> ... </visual>`
2. The other ShapeNode attached to "base" will correspond to the collision geometry defined by `<collision> ... </collision>`

You could remove the `<visual>` element and you'd be left with only one ShapeNode, corresponding to the collision geometry, but then the "base" link will not be visible if you try to visualize the simulation.

Can you copy/paste some example code of how you're retrieving and using the ShapeNodes from "base"? That would allow me to give you some suggestions about how to tweak your code so that you're only handling the collision ShapeNodes.

mxgrey, 2018-04-10:
In case it's helpful, you can do

```
std::vector<dart::dynamics::ShapeNode*> shapeNodes = bodyNode->getShapeNodesWith<dart::dynamics::CollisionAspect>();
```

to get a vector of a BodyNode's ShapeNodes which contain collision data.

EastskyKang, 2018-04-10:
I just realize I can only get collision shape by 

```
body->getShapeNodesWith<dart::dynamics::CollisionAspect>
```

thank you so much for your quick response. I will close this issue.

closed 2018-04-10 by EastskyKang