dartsim/dart #1053: Consider removing FindEIGEN3.cmake

mxgrey, 2018-04-16:
I recently discovered that Eigen3 [provides an Eigen3Config.cmake file](https://eigen.tuxfamily.org/dox/TopicCMakeGuide.html). We should consider this instead of having our own find-module for Eigen.

But one thing to watch out for is it seems that the imported target that gets mentioned in the documentation is not available in Eigen version `3.2.92` (the documentation is for `3.3.4`), so we might want to avoid using the imported target in order to keep supporting older setups.

jslee02, 2018-04-16:
One caveat would be that `Eigen3Config.cmake` is not provided for some Eigen versions that DART supports (>= 3.2.0). We could work around this by using config mode first and then module mode if it fails to find Eigen.

mxgrey, 2018-04-16:
That makes sense if we want to keep supporting older versions of Eigen.

That would also give us the liberty to create imported targets for Eigen for older versions of Eigen that don't automatically provide imported targets.

jslee02, 2019-04-29:
I believe this is resolved by #1212

closed 2019-04-29 by jslee02