dartsim/dart #1056: Proposal: Provide SKIP_<component> cmake variables

mxgrey, 2018-04-25:
It seems that currently we assume that users want to configure and build all dartsim components whose dependencies are available.

Instead, we could provide `SKIP_<component>` variables so that users can decide to skip components that they don't care about, even if the dependency is available, e.g. by calling `cmake .. -DSKIP_planning`. (Alternatively, we could call it `skip_<component>` to avoid mixing cases.)

jslee02, 2018-04-25:
I had the same feeling seeing #1045!

But I would prefer `BUILD_<component>` (or `DART_BUILD_<component>`) **option** by setting the default as `ON`, which I believe it's a more standard way.

mxgrey, 2018-04-25:
Sure, a build option that defaults to on sounds good to me.

mxgrey, 2018-04-26:
There are a lot of possible variable names we could use, so let's do an informal Github poll by putting in reactions (note that you can add multiple reactions):

:+1:: All uppercase (e.g. `BUILD_PLANNING`)
:-1:: All lowercase (e.g. `build_planning`)
:laughing:: Lowercase component name, uppercase everything else (e.g. `BUILD_planning`)

(Ignore case for the following options)
:confetti_ball: (Hooray): `BUILD_<component>`
:confused:: `DART_BUILD_<component>`
:hearts:: `BUILD_DART_<component>`

jslee02, 2018-04-26:
Hm, from my experience, I've seen that uppercase is used for options and global variables, and lowercase is used for local variables.

mxgrey, 2018-04-26:
I think historically cmake has used a lot of uppercase, but it's been transitioning away from that. During that transition, an informal convention arose where uppercase was kept for global variables and for "returned" variables (e.g. `<PACKAGE>_LIBRARIES`, `<PACKAGE>_INCLUDE_DIRS`) while lowercase was used for local variables. I suspect the actual reason for the continued use of uppercase global and returned variables is it's more dangerous to change the case for them, because downstream projects may depend on them.

In fact, new cmake projects are encouraged to potentially use mixed case for returned variables if their package names are lowercase. The latest behavior of [`find_package_handle_standard_args`](https://cmake.org/cmake/help/v3.5/module/FindPackageHandleStandardArgs.html) is to keep the exact case of `<PackageName>` and apply `_FOUND` to it. So if your package name is `dartsim`, it would return `dartsim_FOUND`. The [older behavior](https://cmake.org/cmake/help/v3.0/module/FindPackageHandleStandardArgs.html) was to return `<PACKAGENAME>_FOUND`, where it would convert the package name to uppercase. There are a few other cmake functions that exemplify this.

I think there's a strong trend towards using the original case for components of variables that are referring to literal names of things. This is convenient because it avoids the need to parse and convert between uppercases and lowercases.

psigen, 2018-04-26:
I agree that there is a strong trend towards using the original case for components, but I think this has been very specific to _package_ naming and not to other variables.  This may be related to usage in functions like `FindPackages()` and exported `.CMake` files, because developers want to use exactly consistent casing between the `find` directives and their official library names.

Much of the trend in transitioning away from uppercase that I've see has been in the functions and macros defined in CMake, [possibly because they are the only parts of CMake that are case insensitive in the first place.](https://cmake.org/Wiki/CMake/Language_Syntax#Listfiles_contain_comments_and_commands)

I've cast my vote already above, so don't double count me, but my recommendation is to use `UPPER_CASE` for variable naming, with the specific exception for package names since they have "special" meaning.

mxgrey, 2018-04-27:
Mixed case is fine to me as well; it just looks kind of funny.

jslee02, 2021-04-22:
This is addresed by https://github.com/dartsim/dart/pull/1576. The winner was `DART_BUILD_COMP_<lowercase_component_name>`!

closed 2021-04-22 by jslee02