dartsim/dart #1058: [Question] Is it possible to change constraint solver of dart::simulation::World?

EastskyKang, 2018-04-27:
Hello, 

I am trying to benchmark different constraint solver for various rigid body simulation scenarios. 

Can somebody give me some advice how to change default solver of dart world?

jslee02, 2018-04-27:
DART currently supports two LCP solvers: Dantzig (identical to ODE's `dWorldStep()`) and [Projected Gauss-Seidel](https://en.wikipedia.org/wiki/Gauss%E2%80%93Seidel_method). Here is an example code for changing LCP solver:

```cpp
#include "dart/simulation/simulation.hpp"
#include "dart/constraint/constraint.hpp"

int main()
{
  auto world = dart::simulation::World::create();
  auto cs = world->getConstraintSolver();
  cs->setLCPSolver(dart::common::make_unique<dart::constraint::DantzigLCPSolver>(
      world->getTimeStep()));  // Default LCP solver
  cs->setLCPSolver(dart::common::make_unique<dart::constraint::PGSLCPSolver>(
      world->getTimeStep()));

  return 0;
}
```

EastskyKang, 2018-04-27:
@jslee02 Thank you for your response.

closed 2018-04-27 by EastskyKang