dartsim/dart #1065: [Question] (many) articulated system test failure

EastskyKang, 2018-05-03:
I have built an articulated system simulator using DART sim. 

As I loaded 8 x 8 (64) number of 18-DOF robots on the flat (welded) surface, the contact solving fails as the following video. 

https://youtu.be/imYAui95cgk

(It works properly until 7 x 7 test)

Could you give me any suggestion to fix this problem?

jslee02, 2018-05-03:
It seems that's because the number of contact points in the simulation exceeds the limit, which is 1000 by default. You can check if this is the case by adding following code in you simulation update call:
```cpp
const auto& result = myWorld->getConstraintSolver()->getLastCollisionResult();
const auto numContacts = result.getNumContacts();
if (numContacts > 900)
  std::cout << "Contacts are more than 900\n";
```

You can increase the limit as
```cpp
auto& option = world->getConstraintSolver()->getCollisionOption();
option.maxNumContacts = 5000;
```

If the simulation with increased contact limits is too slow, I recommend reducing the number of contacts by
(1) disabling self-collision if enabled
(2) change the collision detector to `BulletCollisionDetector` if the collision shapes are all primitive shapes.
Just let me know if you're not sure how to do these.

Out of curiosity, how many robots were able to stand on the ground without falling through the ground?

EastskyKang, 2018-05-03:
@jslee02 thank you so much for your response. It works well as I increased maxNumContacts. 

FYI, with 1000 contacts, it works fine with 50 robots and with 5000, at least it works fine with 15 x 15 robots

EastskyKang, 2018-05-03:
@jslee02 would you briefly explain which collision detector is suit for which conditions? I am only using primitives (convex objects and plane) for representing collision body.

EastskyKang, 2018-05-03:
@jslee02 thank you! I'm closing this issue now.

closed 2018-05-03 by EastskyKang