dartsim/dart #1068: LCP internal error, s <= 0 from Dantzig solver

eastskykang, 2018-05-09:
When I simulate the stack of the blocks as the image below, 

[image]

I got the warning message "ODE Message 3: LCP internal error, s <= 0" when I use Dantzig solver. As far as I know, this error is from [ODE Dantzig solver](https://www.ode-wiki.org/wiki/index.php?title=FAQ) (standard solver of Open Dynamics Engine) and it slows down the simulation step.

When I use ODE for simulation, I could tune some parameters such as ERP or CFM, but for Dart, how can I set parameters to attenuate this problem?

mxgrey, 2018-05-09:
We keep a copy of the ODE LCP solver in the DART codebase, so the error will be coming from [here](https://github.com/dartsim/dart/blob/36a6a48aeec4c4e586166c254011d1959ff9f27a/dart/external/odelcpsolver/lcp.cpp#L1007).

I don't think we currently provide any interfaces for tuning the ERP or CFM, but that would be a great feature to add.

mxgrey, 2018-05-10:
One option to consider in the meantime (if you haven't tried it already) is switching to the [PGSLCPSolver](https://github.com/dartsim/dart/blob/master/dart/constraint/PGSLCPSolver.hpp). The PGS method tends to provide more stable results.

You should be able to change the solver by calling

```
world->getConstraintSolver()->setLCPSolver(dart::common::make_unique<dart::constraint::PGSLCPSolver>(world->getTimeStep()))
```

(Disclaimer: That line of code is untested, so it may need some corrections)