dartsim/dart #1079: Unknown collision detector[bullet]

yangyanzhe, 2018-07-10:
#### Environment
DART version: 6.4.0
OS: macOS Sierra 10.13.5

I installed the dart from source following the instructions here: https://dartsim.github.io/install_dart_on_mac.html.

When I run the tutorialBiped-Finished, the messages are shown as follows. Does it matter? I checked bullet and it has already been installed.

```
Warning [Factory-impl.hpp:168] [Factory] Failed to create an object of 'N4dart9collision17CollisionDetectorE' class with the key (type: 'NSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEE'). Returning nullptr instead.
Warning [SkelParser.cpp:753] Unknown collision detector[bullet]. Default collision detector[fcl_mesh] will be loaded.
Warning [Factory-impl.hpp:168] [Factory] Failed to create an object of 'N4dart9collision17CollisionDetectorE' class with the key (type: 'NSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEE'). Returning nullptr instead.
Warning [SkelParser.cpp:753] Unknown collision detector[bullet]. Default collision detector[fcl_mesh] will be loaded.
Warning [SkelParser.cpp:1854] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.
Warning [SkelParser.cpp:1854] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.
Warning [SkelParser.cpp:1854] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.
Warning [SkelParser.cpp:1854] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.
Warning [SkelParser.cpp:1854] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.
Warning [SkelParser.cpp:1854] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.

```

jslee02, 2018-07-10:
It seems Bullet is not correctly detected by CMake. Could you rerun CMake with `-DDART_VERBOSE=ON` (e.g., `cmake .. -DDART_VERBOSE=ON`) and paste the output?

yangyanzhe, 2018-07-10:
Thanks for the prompt reply. Here is the output for 

```
 cmake .. -DART_VERBOSE=ON
```

Output:
```
-- The C compiler identification is AppleClang 9.1.0.9020039
-- The CXX compiler identification is AppleClang 9.1.0.9020039
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- 
-- ============================================
--                 DART 6.4.0
-- ============================================
-- Found EIGEN3: /usr/local/include/eigen3 (found suitable version "3.3.4", minimum required is "3.0.5") 
-- Found CCD: /usr/local/Cellar/libccd/2.0_3/include (found suitable version "2.0", minimum required is "1.4.0") 
-- Found FCL: /usr/local/Cellar/fcl/0.5.0_1/include (found suitable version "0.5.0", minimum required is "0.2.9") 
-- Found ASSIMP: /usr/local/Cellar/assimp/4.1.0/include (found suitable version "4.1.0", minimum required is "3.0.0") 
-- Performing Test ASSIMP_AISCENE_CTOR_DTOR_DEFINED
-- Performing Test ASSIMP_AISCENE_CTOR_DTOR_DEFINED - Success
-- Performing Test ASSIMP_AIMATERIAL_CTOR_DTOR_DEFINED
-- Performing Test ASSIMP_AIMATERIAL_CTOR_DTOR_DEFINED - Success
-- Looking for doxygen - NOT found, to use generating API documentation, please install doxygen
-- Looking for ipopt - NOT found, to use dart-optimizer-ipopt, please install ipopt
-- Performing Test BT_USE_DOUBLE_PRECISION
-- Performing Test BT_USE_DOUBLE_PRECISION - Failed
-- Adding 12 components including the default component 'dart'
-- Adding 42 tests (comprehensive: 15, regression: 4, unit: 23)
CMake Warning at examples/humanJointLimits/InSourceBuild.cmake:11 (find_package):
  By not providing "FindTinyDNN.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TinyDNN", but
  CMake did not find one.

  Could not find a package configuration file provided by "TinyDNN" with any
  of the following names:

    TinyDNNConfig.cmake
    tinydnn-config.cmake

  Add the installation prefix of "TinyDNN" to CMAKE_PREFIX_PATH or set
  "TinyDNN_DIR" to a directory containing one of the above files.  If
  "TinyDNN" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  examples/humanJointLimits/CMakeLists.txt:4 (include)

-- Adding 26 examples
-- Adding 8 tutorials
-- Formatting on 12 source files.
-- Looking for clang-format - NOT found, please install clang-format to enable automatic code formatting
-- 
-- Run 'make' to build all the components
-- Run 'make tests' to build all the unittests
-- Run 'make examples' to build all the examples
-- Run 'make tutorials' to build all the tutorials
-- Run 'make view_docs' to see the API documentation
-- 
-- Configuring done
[... 120 lines cut ...]
  Value of math_headers contained a newline; truncating

CMake Warning:
  Value of optimizer_headers contained a newline; truncating

CMake Warning:
  Value of optimizer_nlopt_headers contained a newline; truncating

CMake Warning:
  Value of planning_headers contained a newline; truncating

CMake Warning:
  Value of simulation_headers contained a newline; truncating

CMake Warning:
  Value of utils_headers contained a newline; truncating

CMake Warning:
  Value of utils_sdf_headers contained a newline; truncating

CMake Warning:
  Value of utils_urdf_headers contained a newline; truncating

-- Build files have been written to: /Users/XXX/project/dart/build
```

jslee02, 2018-07-10:
Hm, that's weird. It seems Bullet is correctly found, but somehow the collision detector factory doesn't work for Bullet. Let me test it on my Mac when I get home.

The warning message says DART falls back to using FCL, which is the default collision detector of DART, instead of the specified Bullet. This shouldn't be a problem except it generally generates more contact points.

jslee02, 2018-07-11:
This should be fixed by #1080. Thanks for the report!

yangyanzhe, 2018-09-18:
**Environment:**
Dart 6.6.1
Cmake 3.12.2
Ubuntu 16.04

Hi, I am installing dart 6.6.1 from source with cmake 3.12.2 on an Ubuntu 16.04, but encounter the same problem:

```
Warning [Factory-impl.hpp:168] [Factory] Failed to create an object of 'N4dart9collision17CollisionDetectorE' class with the key (type: 'NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'). Returning nullptr instead.
Warning [SkelParser.cpp:753] Unknown collision detector[bullet]. Default collision detector[fcl_mesh] will be loaded.
Warning [Factory-impl.hpp:168] [Factory] Failed to create an object of 'N4dart9collision17CollisionDetectorE' class with the key (type: 'NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'). Returning nullptr instead.
Warning [SkelParser.cpp:753] Unknown collision detector[bullet]. Default collision detector[fcl_mesh] will be loaded.
```

But it seems the "collision-bullet" has already been installed in the [CMakeLists.txt](https://github.com/dartsim/dart/blob/v6.6.1/tutorials/tutorialBiped-Finished/CMakeLists.txt). Is there any suggestion on how to resolve it?

yangyanzhe, 2018-09-18:
The log for running 
```
cmake .. -DART_VERBOSE=ON
```
Output:
```
-- ============================================
--                 DART 6.6.1
-- ============================================
-- Looking for doxygen - NOT found, to use generating API documentation, please install doxygen
CMake Warning (dev) at /opt/cmake/cmake-3.12.2-Linux-x86_64/share/cmake-3.12/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  dart/gui/CMakeLists.txt:6 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Adding 13 components including the default component 'dart'
-- Adding 42 tests (comprehensive: 15, regression: 4, unit: 23)
CMake Warning at examples/humanJointLimits/InSourceBuild.cmake:11 (find_package):
  By not providing "FindTinyDNN.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TinyDNN", but
  CMake did not find one.

  Could not find a package configuration file provided by "TinyDNN" with any
  of the following names:

    TinyDNNConfig.cmake
    tinydnn-config.cmake

  Add the installation prefix of "TinyDNN" to CMAKE_PREFIX_PATH or set
  "TinyDNN_DIR" to a directory containing one of the above files.  If
  "TinyDNN" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  examples/humanJointLimits/CMakeLists.txt:4 (include)

-- Adding 26 examples
-- Adding 8 tutorials
-- Formatting on 22 source files.
-- Looking for clang-format - NOT found, please install clang-format to enable automatic code formatting
-- 
-- Run 'make' to build all the components
-- Run 'make tests' to build all the unittests
-- Run 'make examples' to build all the examples
-- Run 'make tutorials' to build all the tutorials
-- Run 'make view_docs' to see the API documentation
-- 
-- Configuring done
CMake Warning:
  Value of collision_bullet_headers contained a newline; truncating

CMake Warning:
  Value of collision_dart_headers contained a newline; truncating

CMake Warning:
  Value of collision_fcl_headers contained a newline; truncating
[... 108 lines cut ...]

CMake Warning:
  Value of optimizer_ipopt_headers contained a newline; truncating

CMake Warning:
  Value of optimizer_nlopt_headers contained a newline; truncating

CMake Warning:
  Value of planning_headers contained a newline; truncating

CMake Warning:
  Value of simulation_headers contained a newline; truncating

CMake Warning:
  Value of utils_headers contained a newline; truncating

CMake Warning:
  Value of utils_sdf_headers contained a newline; truncating

CMake Warning:
  Value of utils_urdf_headers contained a newline; truncating

-- Build files have been written to: /home/XXX/Applications/dart/build

```

jslee02, 2018-09-19:
Which executable are you running?

Btw, you should add `D` between `-` and the CMake option name as:
```
cmake .. -DDART_VERBOSE=ON
```
Note there are two `D`s.

yangyanzhe, 2018-09-19:
Thanks for pointing it out. The output for running:
```
cmake .. -DDART_VERBOSE=ON
```
Output:
```
--
-- ============================================
--                 DART 6.6.1
-- ============================================
--
-- [ Mandatory dependencies for DART core ]
-- Looking for eigen3 - version 3.2.92 found
-- Looking for libccd - version 2.0 found
-- Looking for fcl - version 0.3.2 found
-- Looking for assimp - version 3.2.0 found
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   regex
--   system
--   filesystem
-- Looking for octomap - version 1.6.8 found
-- Looking for PerlModules - NOT found, to colorize gcc messages, please install Regexp::Common Getopt::ArgvFile Getopt::Long Term::ANSIColor (http://www.cpan.org/modules/INSTALL.html)
-- Looking for doxygen - NOT found, to use generating API documentation, please install doxygen
--
-- [ Build summary ]
-- CMAKE_GENERATOR  : Unix Makefiles
-- Compiler ID      : GNU
-- Compiler version : 5.4.0

-- Build type       : Release
-- BUILD_SHARED_LIBS: ON
-- ENABLE_OPENMP    : ON
-- Build gui::osg   : ON
-- Install path     : /usr/local
-- CXX_FLAGS        :  -Wall -Wextra -fPIC -std=c++11
-- CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_SOURCE_DIR : /home/XXX/Applications/dart
-- CMAKE_BINARY_DIR : /home/XXX/Applications/dart/build
-- Looking for ipopt - version 3.11.9 found
-- Looking for nlopt - version 2.4.2 found
-- Looking for ode - version 0.13 found
-- Looking for Bullet - found (single precision)
-- Looking for flann - version 1.8.4 found
-- Looking for tinyxml2 - version 2.2.0 found
-- Looking for urdfdom - 0.4.0 found
-- /home/yanzhe/Applications/dart/build/dart/utils/urdf/BackwardCompatibility.hpp
CMake Warning (dev) at /opt/cmake/cmake-3.12.2-Linux-x86_64/share/cmake-3.12/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  dart/gui/CMakeLists.txt:6 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for OpenGL - version  found
-- Looking for freeglut3 - version  found
-- Looking for OpenSceneGraph - 3.2.1 found
[... 241 lines cut ...]

CMake Warning:
  Value of optimizer_headers contained a newline; truncating

CMake Warning:
  Value of optimizer_ipopt_headers contained a newline; truncating

CMake Warning:
  Value of optimizer_nlopt_headers contained a newline; truncating

CMake Warning:
  Value of planning_headers contained a newline; truncating

CMake Warning:
  Value of simulation_headers contained a newline; truncating

CMake Warning:
  Value of utils_headers contained a newline; truncating

CMake Warning:
  Value of utils_sdf_headers contained a newline; truncating

CMake Warning:
  Value of utils_urdf_headers contained a newline; truncating
```

jslee02, 2018-09-21:
Thanks! It seems bullet is detected and built with DART. But I'm still wondering which executable (example) are you running. Whichever it is, please make sure `dart-collision-bullet` target is linked to the executable.

yangyanzhe, 2018-09-21:
> Thanks! It seems bullet is detected and built with DART. But I'm still wondering which executable (example) are you running. Whichever it is, please make sure `dart-collision-bullet` target is linked to the executable.

It is tutorialBiped-Finished. It seems the "collision-bullet" has already been stated as a dependency in the [CMakeLists.txt](https://github.com/dartsim/dart/blob/v6.6.1/tutorials/tutorialBiped-Finished/CMakeLists.txt). I also tried to change ${DART_LIBRARIES} to be 
```dart dart-collision-bullet dart-utils-urdf dart-gui```. That doesn't help as well. That is weird to me actually.

jslee02, 2018-09-21:
I see. It seems the current Factory implementation doesn't work as expected. This needs to be fixed.

Could you use the following code to change the collision detector in the meantime?
```cpp
world->getConstraintSolver()->setCollisionDetector(
      std::make_shared<dart::collision::BulletCollisionDetector>());
```

yangyanzhe, 2018-09-21:
Thanks! I compiled it with adding 
```c++
#include <dart/collision/bullet/BulletCollisionDetector.hpp>
```
at the header and moving the constructer of BulletCollisionDetector from protected to public.

closed 2018-09-21 by yangyanzhe