dartsim/dart #1085: Running osgWamIkFast example

junhyeokahn, 2018-07-17:
Hello,
I am currently trying an osgWamIkFast example.
Once I run the code, I got warnings below.
```
Warning [SharedLibraryManager.cpp:50] [SharedLibraryManager::load] The given path doesn't exist. Returning nullptr.
Error [SharedLibraryIkFast.cpp:95] [SharedLibraryIkFast] Could not load dynamic library 'libexample_wamIk.dylib'. This SharedLibraryIkFast is invalid.
Warning [IkFast.cpp:225] [IkFast::getDofs] This analytical IK was not able to configure properly, so it will not be able to compute solutions. Returning an empty list of dofs.
```
Where could I find the library? Should the library be preinstalled?

jslee02, 2018-07-17:
The library should be built along with `osgWamIkFast` automatically by the CMake script. The problem is that library isn't copied to the same location where `osgWamIkFast` copied, which is a bug.

Let me fix this soon. Basically adding the following code [here](https://github.com/dartsim/dart/blob/release-6.5/examples/osgExamples/osgWamIkFast/InSourceBuild.cmake#L15) should fix the problem:
```cmake
set_target_properties(example_wamIk
  PROPERTIES
    LIBRARY_OUTPUT_DIRECTORY "${CMAKE_BINARY_DIR}/bin"
)
```

junhyeokahn, 2018-07-17:
Thanks for immediate reply.
Is the way you generate ```ikfast71.Transform6D.4_6_9_10_11_12_f8.cpp``` to use ```ikfast``` module from Openrave?

jslee02, 2018-07-17:
> Is the way you generate ikfast71.Transform6D.4_6_9_10_11_12_f8.cpp to use ikfast module from Openrave? (e.g. ikfast.py)

Correct!

junhyeokahn, 2018-07-17:
Thank you so much. I need to spend some time to install Openrave and run it :) I will close the issue.

jslee02, 2018-07-19:
Resolved by #1087

PR #1087 (2018-07-18): Fix osgWamIkFast build
Fixes #1085 

***

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
closed 2018-07-19 by jslee02