dartsim/dart #1086: Non-deterministic simulation

yangyanzhe, 2018-07-17:
Help needed about simulation status

Description: The following function will return different results occasionally in multiple runs. Is it normal? From my understanding, the state of the engine is (pos, vel, acc, forces), right? Do I miss some variables containing the simulation information?

```
MyState forward(WorldPtr & _world, const MyState & start, const Eigen::VectorXd & torque)
{
     MyState end;

    _world->reset();
    auto _biped = _world->getSkeleton("biped");
    _biped->setPositions(start.qpos);
    _biped->setVelocities(start.qvel);
    _biped->setAccelerations(start.qacc);
    _biped->setForces(torque);

    int numStep = static_cast<int>(round(1.0f / (_world->getTimeStep() * samplePerSecond)));
    for(int i = 0; i < numStep; i++)
    {
        _world->step();
    }

    end.qpos = _biped->getPositions();
    end.qvel = _biped->getVelocities();
    end.qacc = _biped->getAccelerations();
    return end;
}
```

jslee02, 2018-07-18:
Pos, vel, and force should be sufficient. Note that acc will be computed (by forward dynamics computation) given pos/vel/force.

To be clear, do you have any other skeleton in the world that could interact with `biped`?

Also note that the forces are reset to zero after every world step (i.e., `step()`) so you should set the force for every time step. For example,
```cpp
for(...)
{
  biped->setForces(torque);
  _world->step();
```
or pass `false` to `step()` to keep passing the same force (see [this comment](https://github.com/dartsim/dart/blob/master/dart/simulation/World.hpp#L208-L211)).

yangyanzhe, 2018-07-18:
There is a floor(like the one in the [biped tutorial](https://dartsim.github.io/tutorials_biped.html)) to interact with the skeleton.

yangyanzhe, 2018-08-01:
Code and data to reproduce the problem: [repo](https://github.com/yangyanzhe/TestDeterministic)

mxgrey, 2018-08-02:
Thanks for the test code, it was very helpful for identifying the issue.

It turns out that the "non-deterministic" factor was a state outside of position and velocity: the constraint impulse. In [this line](https://github.com/yangyanzhe/TestDeterministic/blob/master/Controller.h#L107) you're using `mBiped->getConstraintForces()`, which is not a function of position or velocity, so it does not get reset by simply setting the positions and velocities.  If you just remove `+ mBiped->getConstraintForces()` from that line, you'll see that the results are much more consistent. Tests 23, 25, and 26 still seem to be failing, though.

I would guess those three failures may be due to some statefulness in the underlying collision detector or constraint solver. It seems that the easiest way to get complete "determinism" from the simulation is to add `InitEnv();` on [this line](https://github.com/yangyanzhe/TestDeterministic/blob/master/TestDeterministic.cpp#L208). That way, you're starting with a fresh environment in each test, and there's no tricky statefulness hiding behind the scenes in the simulation environment.

All that being said, it would definitely be a good idea for us to examine how to make these "hidden" simulation states more transparent. We accomplished this for dynamics states using the `State` and `Properties` mechanisms, so maybe we should redesign the world, collision detection wrappers, and constraint solvers to use a similar scheme.

I'll close this issue and open a new one for us to discuss design and implementation changes.

closed 2018-08-02 by mxgrey

yangyanzhe, 2018-08-02:
Is there a way to reset data structure "State" for the skeleton? I found if I clone the world, they will succeed as well. From the [comment](https://github.com/dartsim/dart/blob/release-6.4/dart/simulation/World.hpp#L98) of the world->clone, it seems dropping the information stored in the state of the skeleton helps?

yangyanzhe, 2018-08-02:
Just a follow-up on using InitEnv() to help start with a fresh environment. I tried three options to reset the environment, 

```
        /// success
        /// option 1: init env, i.e. recreate the environment
        // InitEnv();

        /// success
        /// option 2: clone the world and use the new world
        auto worldCopy = mWorld->clone();
        mWorld = worldCopy;
        mBiped = mWorld->getSkeleton("biped");
        mController->setSkeleton(mBiped);

        /// fail
        /// option 3: reset state, the initStates are store before test using getState
        //mBiped->setState(initBipedState);
        //mFloor->setState(initFloorState);
```

mxgrey, 2018-08-03:
In principle, option 3 ought to work, except it seems we're neglecting a few members of the joint state. I've made a comment about fixing it [here](https://github.com/dartsim/dart/issues/1089#issuecomment-410126649).