dartsim/dart #1091: Build on Ubuntu 18.04 (bionic) fails

xemjeff, 2018-08-03:
I am not able to get a build of the examples files on U18.04 bionic. I am building outside of the source directory, which is in /usr/share/doc/dart/examples (owned by root).

using cmake v 3.10.2

Here are the steps I've taken:
```
% sudo apt-get install libdart6-all-dev
% cd ~/code/dart
% md build
% cd build
% cmake /usr/share/doc/dart/examples
```
I get the following error:

```
jeff@dragon-vm:~/code/dartex/build$ cmake /usr/share/doc/dart/examples
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.2.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
CMake Error at simpleFrames/CMakeLists.txt:17 (add_executable):
  add_executable called with incorrect number of arguments

CMake Error at simpleFrames/CMakeLists.txt:19 (target_link_libraries):
  Cannot specify link libraries for target "simpleFrames" which is not built
  by this project.

-- Configuring incomplete, errors occurred!
See also "/home/jeff/code/dartex/build/CMakeFiles/CMakeOutput.log".
```
Any suggestions as to what I might be doing wrong are appreciated.

jslee02, 2018-08-04:
It seems CMake fails to find the source files for some reason.

Can you replace [these lines](https://github.com/dartsim/dart/blob/release-6.5/examples/simpleFrames/CMakeLists.txt#L16-L17) with followings and share the output?
```cmake
file(GLOB srcs "*.cpp" "*.hpp")
add_executable(${PROJECT_NAME} ${srcs})
message(STATUS "[DEBUG] PROJECT_NAME: ${PROJECT_NAME}")
message(STATUS "[DEBUG] srcs: ${srcs}")
```

xemjeff, 2018-08-04:
yes, can't seem to find the srcs.

here's the output.
the examples are installed in /`usr/share/doc/dart/examples`, and all owned by root

```
jeff@dragon-vm:~/code/dartex/build$ cmake /usr/share/doc/dart/examples
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.2.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
-- Found DART: /usr/include (Required is at least version "6.5.0") found components:  utils-urdf gui dart 
CMake Error at simpleFrames/CMakeLists.txt:17 (add_executable):
  add_executable called with incorrect number of arguments

-- [DEBUG] PROJECT_NAME: simpleFrames
-- [DEBUG] srcs: 
CMake Error at simpleFrames/CMakeLists.txt:21 (target_link_libraries):
  Cannot specify link libraries for target "simpleFrames" which is not built
  by this project.

-- Configuring incomplete, errors occurred!
See also "/home/jeff/code/dartex/build/CMakeFiles/CMakeOutput.log".

```

xemjeff, 2018-08-05:
I am progressing along, leaving out the `humanJointsLimit`

One problem that exists in many of the examples is that the some of the source (.cpp, .hpp) is stiill gzipped, and then cmake cannot find the source files.  I remedied this by running `gunzip -r *` but it appears that some step is missing. (or cmake magic is needed to look inside the .gz files for source.).

jslee02, 2018-08-06:
The issue of resource retriever should be resolved by #1095. Could you please upgrade DART by:
```
sudo apt update
sudo apt upgrade libdart6-all-dev
```
and let us know if the problem is still persistent? Thanks!

jslee02, 2018-08-08:
I believe this is resolved. Feel free to reopen if not.

PR #1094 (2018-08-05): Fix example build 6.6
This PR partially fixes #1091.
PR #1095 (2018-08-06): Add option to DartResourceRetriever to search from environment variable DART_DATA_PATH
This PR adds a fallback option to `DartResourceRetriever` to use an environment variable to search DART data files. The existing preprocessor, `DART_DATA_GLOBAL_PATH`, doesn't work if the package manager uses temporary install directory (e.g., Launchpad PPA and Homebrew).

This should resolve #1091 and https://github.com/dartsim/homebrew-dart/issues/54.

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull re [... cut]
closed 2018-08-08 by jslee02