dartsim/dart #1096: Clang emits -Wfloat-equal warning in consumer projects

mxgrey, 2018-08-06:
I'm using clang-6.0 on Ubuntu 16.04, and I've found that projects that consume dartsim version 6 will get hit by the `-Wfloat-equal` warning from `GenericJoint.hpp` if they try to instantiate any joint types.

The consumer projects can avoid this by adding the dartsim headers as `SYSTEM` headers, but if it's possible to do a bitwise comparison of floating point variables without triggering this warning, then it would be good to switch to that.

The only way I can think of doing that off the top of my header is something goofy like:

```
reinterpret_cast<long long&>(value) == reinterpret_cast<long long&>(Base::mAspectProperties.mField)
```

but I'd prefer to avoid solutions that involve `reinterpret_cast` or make assumptions about the sizes of primitive types.

jslee02, 2018-08-06:
Could you let me know which line is it in `GenericJoint.hpp`?

mxgrey, 2018-08-06:
It's [this line](https://github.com/dartsim/dart/blob/release-6.7/dart/dynamics/detail/GenericJoint.hpp#L70) in `detail/GenericJoint.hpp`, where we compare the user-requested value against the existing value.

It's a rare case where we really do want bit-wise comparison of two floats.

jslee02, 2018-08-06:
If we just want to suppress warnings, then we could just disable that warning for the comparison using [this approach](GenericJoint_SET_IF_DIFFERENT).

I think the next question is that whether we really want to do bit-wise comparison for float point values, which I don't have a preference right now. :thinking:

jslee02, 2018-08-06:
I agree. Let's then perform the bitwise comparison. FYI, I found a [SO post](https://stackoverflow.com/a/1723938/3122234) about size agnostic bitwise comparison.

mxgrey, 2018-08-06:
Yes, the equality operation should be the kind of comparison we want. The post you linked seems to be talking about comparing specific bits within the floating point representation, but we only really care about comparing the exact values as a whole.