dartsim/dart #1105: v6.6.1 SHA256 shift.

DomT4, 2018-08-26:
Hey folks,

Apologies for erasing the issue template but I'm not really reporting a bug, as such, so I wasn't sure there was anything relevant there.

Homebrew merged v6.6.1 in https://github.com/Homebrew/homebrew-core/pull/30979, during which our CI checked & confirmed the SHA256 of the archive tarball (https://github.com/dartsim/dart/archive/v6.6.1.tar.gz) was `4000ad91dc20a5fee89f2252c9eb28cef62c89df88bff218374b70f38245a61c`. 12 days later the formula's archive tarball was put back through CI on an unrelated issue and popped a SHA256 of `86cc3249938602754f773e0843f415c290bd2608729ab3e219de78f90bdd4d6b`.

Can y'all confirm if there was a retag there? It's possible Homebrew jumped the gun; I notice v6.6.1 hasn't been tagged as the latest release here yet.

jslee02, 2018-08-26:
It seems Homebrew/homebrew-core#30979, which I haven't noticed, was created before I re-tagged v6.6.1. I had not released v6.6.1 because it wasn't stable yet.

As v6.6.1 is now stable, let me release it and submit a PR to homebrew-core to update the SHA256.

DomT4, 2018-08-26:
Fixed via https://github.com/Homebrew/homebrew-core/pull/31480. Thanks again 🍻

closed 2018-08-26 by DomT4

jslee02, 2018-08-26:
Thanks for sharing this issue! :tada: