dartsim/dart #1108: velocity profile jump when collision exists

junhyeokahn, 2018-09-03:
Hello,
I have observed velocity jump when the contact exists, which prevent me to use high gains at velocity.
For example, in the example of ```osgAtlasSimbicon```, the figure below is the graph of num_time_step vs joint velocity at 31st joint of Atlas.
[image]
[image]

It is natural having velocity jump during the impact event.
However, there seems to be a velocity discontinuity even in non-impact situation.
Probably I am asking this because I do not understand how the simulation works behind the scene.

jslee02, 2018-09-05:
Quick thought: Could you check if there is no other constraint forces applied to the skeleton using [`Skeleton::getConstraintForces()`](https://github.com/dartsim/dart/blob/master/dart/dynamics/Skeleton.hpp#L868)? If there is no constraints, it should return a zero vector.

junhyeokahn, 2018-09-05:
Thanks for your comment.
I just check ```getConstraintForces()``` and I was able to observe there are some constraint forces at the moment velocity jumps.

Could you explain what causes constraint forces?
In my simulation, I just fall my humanoid robot down with zero torque input. It has collision objects only at the both foots and contacting with ground.

jslee02, 2018-09-07:
Thanks for sharing the test results with the nice graphs! Besides contact constraints, there could be other constraints such as joint limits, joint Coulomb friction, servo motor, dynamic joints, and so on.  Based on your graphs, my best guess is joint friction. Could you quickly set all the joint friction coefficients to zeros by using `Joint::setCoulombFriction()`?

By the way, the velocity profile and constraint force profile look reasonable, but the position profile looks a bit odd to me because it's smooth even with the velocity jumps. :thinking: Do you also see the velocity jumps visually in the simulation?

junhyeokahn, 2018-09-07:
Thanks for your suggestion.
I checked the same thing with zero joint friction but I had same result.
However, I set joint damping in the ```urdf``` file from 0.5 to 0 and now it shows smoother profile as below.
[image]
Now it seems very smooth but there is still little peak on the constraint forces. I think I could neglect it but just for curiosity.

jslee02, 2018-09-19:
Could you check out [this function](https://github.com/dartsim/dart/blob/master/dart/constraint/ConstraintSolver.cpp#L386-L536)? It's where all the constraints are generated. You should be able to see if no constraints are generated actually.

jslee02, 2018-09-21:
> Do you think it may be the reason for the velocity jump?

It's possible, but not the main reason. My best guess is that the velocity jump happens because there are tiny bouncing between the colliding bodies at the contact point.

In DART, constraint errors are resolved in velocity level using impulses. Due to the discrete time collision checking, we cannot avoid penetration, and the penetration is also handled using impulse so that the impulse pushes the bodies to remove the penetration. Here some artificial bouncing can happen. We can teak the effect by changing some relevant parameters.

Regarding the time-varying contact numbers, I would recommend using other collision detectors than FCL. FCL returns more than sufficient contact numbers because we use mesh shape that is corresponding the original primitive shape, because the primitive collision of FCL isn't complete yet. We plan to address many contact point problem, but it hasn't been addressed yet.

junhyeokahn, 2018-09-21:
Thanks for your kindness :)
One thing weird though is the smoothness of position under velocity jumps.
Those velocities were achieved by ```getvelocities()``` in ```Skeleton```

jslee02, 2018-09-21:
I had the same impression, but it now seems to make sense to me. Although the velocity jump visually large, those jumps only happens one or several times with some interval. I think the impact (its integration) is negligible. Assuming your timestep is 1e-3, the integration of the jump part (vel - avg_vel_around_the_jump) in the position profile would be the slope change of -2-e3 (= -2 * 1e-3). If you zoom in the position graph further, I guess you would see the non-smoothness.

junhyeokahn, 2018-09-23:
Sounds reasonable.
I appreciate all your help!

jslee02, 2018-09-23:
Hope my answers helped! I'm closing this issue, but feel free to reopen if you have a further issue.

closed 2018-09-23 by jslee02