dartsim/dart #1113: Error Building DART 6.6 with Octomaps

kamatajinkya, 2018-09-19:
DART version: 6.6.1
OS: Ubuntu 14.04
Complier: GNU 4.8.4

DART fails to build with the following error.
```
CMakeFiles/dart.dir/dynamics/VoxelGridShape.cpp.o: In function `dart::dynamics::VoxelGridShape::VoxelGridShape(double)':
VoxelGridShape.cpp:(.text+0xa97): undefined reference to `octomap::OcTree::OcTree(double)'
CMakeFiles/dart.dir/dynamics/VoxelGridShape.cpp.o: In function `dart::dynamics::VoxelGridShape::VoxelGridShape(double)':
VoxelGridShape.cpp:(.text+0xc9c): undefined reference to `octomap::OcTree::OcTree(double)'
CMakeFiles/dart.dir/dynamics/VoxelGridShape.cpp.o: In function `dart::dynamics::VoxelGridShape::VoxelGridShape(std::shared_ptr<octomap::OcTree>)':
VoxelGridShape.cpp:(.text+0xf7b): undefined reference to `octomap::OcTree::OcTree(double)'
CMakeFiles/dart.dir/dynamics/VoxelGridShape.cpp.o: In function `dart::dynamics::VoxelGridShape::VoxelGridShape(std::shared_ptr<octomap::OcTree>)':
VoxelGridShape.cpp:(.text+0x1333): undefined reference to `octomap::OcTree::OcTree(double)'
collect2: error: ld returned 1 exit status
make[2]: *** [lib/libdart.so.6.6.1] Error 1
make[1]: *** [dart/CMakeFiles/dart.dir/all] Error 2
make: *** [all] Error 2
```
#### Steps to Reproduce

1. Clone dart
2. Create and open build directory
3. Run cmake and make

Terminal was sourced with ROS Indigo setup. octomap_ros was installed on machine. The build does not fail if octomap_ros is not installed.

mxgrey, 2018-09-21:
Thanks for the report.

It looks like between octree version [1.6.4](https://github.com/OctoMap/octomap/blob/v1.6.4/octomap/include/octomap/OcTree.h) and [1.7.1](https://github.com/OctoMap/octomap/blob/v1.7.1/octomap/include/octomap/OcTree.h), the definition of the `OcTree(double)` constructor went from being inlined to being a symbol exported by the library.

It looks like the ROS-installed [octomap](http://wiki.ros.org/octomap) uses v1.6.4 for the Indigo release, where the definition of the constructor is inlined. Hypothetically, if you have a newer version of Octomap installed on your system, then it's possible that the compiler is choosing to use the system-installed headers (which will simply be in `/usr/include`) while linking against the octomap library that is shipped with ROS-Indigo. If your system-installed version of Octomap is version 1.7.1 or later, then the header will not contain the constructor definition, but the ROS-Indigo octomap library will *also* not contain the constructor definition, resulting in an undefined reference.

My recommendation is to not have ROS Indigo sourced while trying to build DART.

Although since octomap doesn't version its ABI, if you need to use DART in a ROS node which also depends on the ROS-Indigo octomap, then the symbols of the two different library versions could collide and result in undesirable behavior (which may include crashing the node). In that situation, I would recommend uninstalling octomap from your system (probably using `sudo apt remove liboctomap-dev`) and rely solely on the version of octomap distributed with ROS-Indigo (in which case, you'll *need* to have ROS-Indigo sourced when you build DART).

Let me know if anything about my guesses are wrong (e.g. if you don't have a different version of Octomap installed on your system).

kamatajinkya, 2018-09-21:
@jslee02 
I have added the output of cmake with verbose option here.
It is using octomap 1.8.0 which I had installed previously.
If I remove octomap library or source ROS-Indigo environment it is indeed using octomap 1.6.4.
I am planning to use dart with ROS node and as such need to source the environment. I have seen that ROS-Kinetic uses octomap 1.7.1. Do you suggest that I use that instead of Indigo?
[cmake-dart.log](https://github.com/dartsim/dart/files/2407077/cmake-dart.log)

jslee02, 2018-09-23:
I also agree with @mxgrey's suggestion:
> In that situation, I would recommend uninstalling octomap from your system (probably using sudo apt remove liboctomap-dev) and rely solely on the version of octomap distributed with ROS-Indigo (in which case, you'll need to have ROS-Indigo sourced when you build DART).

Adding to it, I would recommend using ROS-Kinect, which is the recommended version for Ubuntu Xenial. IIRC, our SW stack start to rely on Xenial so it would be a better choice. Also, I remember I tested voxel map (DART + AIKIDO) against that combination.

jslee02, 2019-04-24:
Closing due to lack of response. Please feel free to reopen if the issue persists.

closed 2019-04-24 by jslee02