dartsim/dart #1114: packages.osrfoundation.org to import DART packages from ppa

j-rivero, 2018-09-24:
Dear DART developers:

For different works at Open Robotics and to deliver a complete solution in just one repo for our .deb packages users, we plan on importing the last rounds of dart6 .deb packages available in your PPA.

Let me know if you see any problem with it.
Thanks.

jslee02, 2018-09-24:
This sounds a good plan to me. As I don't have experience with deb servers, I have some novice questions:

* Would it be there some conflicts when a user added our PPA and the OSRF deb server?
* Is it simple copying the deb files from our PPA to the OSRF deb server or it requires additional work?

j-rivero, 2018-09-24:
>     * Would it be there some conflicts when a user added our PPA and the OSRF deb server?

It should not be any kind of conflict since we are providing the same packages with exactly the same versions and metadata. Usually the package manager check some user configurations that define priority between sources (if they exist) and if the user does not provide any setup on which sources to use it will look for the highest version available. In our case it is exactly the same so my impression is that it is up to the package manager to pick one of both arbitrarily.  

>     * Is it simple copying the deb files from our PPA to the OSRF deb server or it requires additional work?

Repos are a bit more complex that just a file directory, in our case we use one of the different .deb repo managers. In any case the actions needed are only in our server side.

Thanks Jeongseok.

closed 2018-09-24 by j-rivero

jslee02, 2018-09-24:
Thank you for the explanation and having DART packages in the server!