dartsim/dart #1180: AppVeyor fails to build DART on Windows

jslee02, 2018-10-23:
Currently, `master` and `release-6.7` branch aren't built successfully. The last successful build was [this](https://ci.appveyor.com/project/jslee02/dart/builds/18361637).

jslee02, 2018-11-10:
It appears that the build issue is related to this [bug](https://developercommunity.visualstudio.com/content/problem/313306/vs2017-158-internal-compiler-error-msc1cpp-line-15-1.html) and will be fixed in VS 16.0.

scpeters, 2019-04-02:
VS 2019 has been released:

* https://docs.microsoft.com/en-us/visualstudio/releases/2019/release-notes

jslee02, 2019-04-03:
Thanks for the heads up! But [Appveyor's Windows image doesn't support CMake 3.14](https://www.appveyor.com/updates/2019/02/11/) yet that includes the [VS2019 generator](https://cmake.org/cmake/help/v3.14/release/3.14.html#id4).

scpeters, 2019-04-08:
I've been trying to build dart in a windows10 VM with VS 2019 and cmake 3.14.1, but I'm getting some cmake errors:

~~~
-- ============================================
--                 DART 6.8.0
-- ============================================
-- Found ccd: C:/Users/scpet/Desktop/vcpkg/installed/x64-windows/include (Required is at least version "2.0")
CMake Error at cmake/DARTFindccd.cmake:15 (set_target_properties):
Property INTERFACE_LINK_LIBRARIES may not contain link-type keyword
"optimized".  The INTERFACE_LINK_LIBRARIES property may contain
configuration-sensitive generator-expressions which may be used to specify
per-configuration rules.
Call Stack (most recent call first):
cmake/DARTMacros.cmake:81 (include)
cmake/DARTFindDependencies.cmake:17 (dart_find_package)
CMakeLists.txt:144 (include)

-- Found fcl: C:/Users/scpet/Desktop/vcpkg/installed/x64-windows/include (Required is at least version "0.3.2")
CMake Error at cmake/DARTFindfcl.cmake:16 (set_target_properties):
Property INTERFACE_LINK_LIBRARIES may not contain link-type keyword
"optimized".  The INTERFACE_LINK_LIBRARIES property may contain
configuration-sensitive generator-expressions which may be used to specify
per-configuration rules.
Call Stack (most recent call first):
cmake/DARTMacros.cmake:81 (include)
cmake/DARTFindDependencies.cmake:21 (dart_find_package)
CMakeLists.txt:144 (include)

-- Found assimp: C:/Users/scpet/Desktop/vcpkg/installed/x64-windows/include
-- Found Assimp , but Assimp >= 3.2 is required
CMake Error at cmake/DARTFindassimp.cmake:23 (set_target_properties):
Property INTERFACE_LINK_LIBRARIES may not contain link-type keyword
"optimized".  The INTERFACE_LINK_LIBRARIES property may contain
configuration-sensitive generator-expressions which may be used to specify
per-configuration rules.
Call Stack (most recent call first):
cmake/DARTMacros.cmake:81 (include)
cmake/DARTFindDependencies.cmake:25 (dart_find_package)
CMakeLists.txt:144 (include)

-- Performing Test ASSIMP_AISCENE_CTOR_DTOR_DEFINED
CMake Error at C:/Users/scpet/Desktop/ws/dartsim/src/dart/build/CMakeFiles/CMakeTmp/cmTC_321d3Targets.cmake:18 (set_target_properties):
Property INTERFACE_LINK_LIBRARIES may not contain link-type keyword
"optimized".  The INTERFACE_LINK_LIBRARIES property may contain
configuration-sensitive generator-expressions which may be used to specify
per-configuration rules.
Call Stack (most recent call first):
C:/Users/scpet/Desktop/ws/dartsim/src/dart/build/CMakeFiles/CMakeTmp/CMakeLists.txt:13 (include)

CMake Error at C:/Program Files/CMake/share/cmake-3.14/Modules/CheckCXXSourceCompiles.cmake:110 (try_compile):
Failed to configure test project build system.
Call Stack (most recent call first):
cmake/DARTFindDependencies.cmake:39 (check_cxx_source_compiles)
CMakeLists.txt:144 (include)

-- Configuring incomplete, errors occurred!
~~~

I tried tweaking some of the find modules, and managed to confirm some of the lines of code that are causing the problem:

~~~
diff --git a/cmake/Findassimp.cmake b/cmake/Findassimp.cmake
index 7300ddc8b..fd837fdb2 100644
--- a/cmake/Findassimp.cmake
+++ b/cmake/Findassimp.cmake
@@ -26,7 +26,7 @@ find_path(ASSIMP_INCLUDE_DIRS assimp/scene.h
 
# Libraries
if(MSVC)
-  set(ASSIMP_LIBRARIES optimized assimp debug assimpd)
+  set(ASSIMP_LIBRARIES assimp)
else()
find_library(ASSIMP_LIBRARIES
NAMES assimp
diff --git a/cmake/Findccd.cmake b/cmake/Findccd.cmake
index 23f832306..18ac4f885 100644
--- a/cmake/Findccd.cmake
+++ b/cmake/Findccd.cmake
@@ -43,7 +43,7 @@ if(NOT CCD_FOUND)
 
# Libraries
if(MSVC)
-    set(CCD_LIBRARIES optimized ccd debug ccdd)
+    set(CCD_LIBRARIES ccd)
else()
# Give explicit precedence to ${PC_CCD_LIBDIR}
find_library(CCD_LIBRARIES
diff --git a/cmake/Findfcl.cmake b/cmake/Findfcl.cmake
index ccc6c8db7..9c4cbe6df 100644
--- a/cmake/Findfcl.cmake
+++ b/cmake/Findfcl.cmake
@@ -37,7 +37,7 @@ endif()
 
# Libraries
if(MSVC)
-  set(FCL_LIBRARIES optimized fcl debug fcld)
+  set(FCL_LIBRARIES fcl)
else()
# Give explicit precedence to ${PC_FCL_LIBDIR}
find_library(FCL_LIBRARIES
 
~~~

scpeters, 2019-04-08:
that errors isn't the only blocker; there is also an upstream issue with boost-system and boost-filesystem: https://github.com/Microsoft/vcpkg/issues/5807

jslee02, 2019-04-08:
Thanks for the investigation! Let me apply your patch to [`vs2019`](https://github.com/dartsim/dart/tree/vs2019) branch (still CMake 3.14 is not available in the CI yet).

scpeters, 2019-04-08:
My patch might break the windows debug builds, so I'm not sure it's a good idea. I was mainly trying to identify the problematic lines of code. I don't know what's the right cmake way to specify debug library names.

scpeters, 2019-04-09:
VS 2019 will require a new version of boost (1.70): https://github.com/boostorg/build/issues/389#issuecomment-478682984

scpeters, 2019-04-16:
I have been able to get boost building with VS 2019 using a modification to boost-build mentioned in https://github.com/Microsoft/vcpkg/issues/5976#issuecomment-482966168 and dart is successfully configuring, but I'm seeing a bunch of compilation errors. I've pasted a truncated list of them into the following gist:

* https://gist.github.com/scpeters/8adfb1d2c9eb560ce6582844a0887363

A recurring them is complaining about the `<<` operator in Console.hpp:

~~~
dart/common/detail/EmbeddedAspect.hpp(174): error C2593: 'operator <<' is ambiguous [build\dart\dart.vcxproj]
  dart/common/Console.hpp(67): note: could be 'unknown-type dart::operator <<<char[63]>(std::ostream &,const T (&))'
          with
          [
              T=char [63]
          ]
~~~

I don't know if there are some compiler flags missing that would explain this?

scpeters, 2019-04-16:
Also, I was just peeking at the appveyor builds, which have still been running, and it is also failing with the `INTERFACE_LINK_LIBRARIES may not contain link-type keyword "optimized"` message that I am seeing.

I think that issue is from #1212. I also noticed that [appveyor build 6085](https://ci.appveyor.com/project/jslee02/dart/builds/20832922/job/bd0bnggttoewe2cr) from December has the internal compiler error, while [build 6086](https://ci.appveyor.com/project/jslee02/dart/builds/22780162/job/0t3lhsu9yctwa9c8) from March has the cmake configuration error.

jslee02, 2019-04-16:
> A recurring them is complaining about the `<<` operator in Console.hpp:

This should be fixed by [this change](https://github.com/dartsim/dart/compare/vs2019#diff-af3b638bc2a3e6c650974192a53c7291R169), which is included in `vs2019` branch.

scpeters, 2019-04-16:
Oh, thanks for the tip! I'm trying to rebuild with that now. I also pushed in a7d67a00923311684e1328c21042306513f229a3 a fix to the cmake generator and toolset flags based on the [cmake docs](https://cmake.org/cmake/help/git-stage/generator/Visual%20Studio%2016%202019.html)

scpeters, 2019-04-17:
azure-pipelines also don't have cmake 3.14 yet https://github.com/Microsoft/azure-pipelines-image-generation/issues/754

scpeters, 2019-04-21:
I've tried fixing the cmake error using generator expressions in 73ef8c2

I've also set the windows vmimage to windows-2019, which has cmake 3.14

jslee02, 2022-05-30:
I haven't tracked all the changes for this, but I believe building tests, examples, and tutorials on Windows now works as of https://github.com/dartsim/dart/commit/22bbcde5ba645fb4c799dd0abd625b2ae8105835

closed 2022-05-30 by jslee02