dartsim/dart #1183: Tips for making simulations faster

costashatz, 2018-10-24:
This might be related to #1068, but I would like a few tips to make my simulations run faster. To make my question more specific:

- I have a simulation with a [YuMi robot](https://new.abb.com/products/robotics/industrial-robots/yumi) and some small parts for manipulation (mesh objects)
- I need the time-step of the simulation to be small (i.e., <=0.001) because the parts for manipulation are quite small and if I increase the simulation step I get weird collision responses
- What things can I do to make it run faster? At the moment, I get quite big times. Should I change the constraint solver?

Overall I would like to know what I can do to make it faster (I have already tried to have coarse meshes for collision and it does improve a bit, but not as much as I would like to)... Thanks!

jslee02, 2018-10-24:
It appears the number of contact points has some impact based on your experiments (using coarse meshes). In this case, you might want to try using bullet collision detection. This is because FCL returns all the contact points of the triangles in contact while bullet only returns up to 4 contacts per collision object pair.

costashatz, 2018-10-24:
> In this case, you might want to try using bullet collision detection.

Thanks for the fast reply. Changing to Bullet made everything much faster, but self collisions are not being processed (i.e., collisions from the same object are being ignored!). What could be the cause for this? Any ideas?

jslee02, 2018-10-24:
Thanks for the answers! It seems self-collision with bullet has some issues. Let me investigate further late today.

costashatz, 2018-10-24:
> Thanks for the answers! It seems self-collision with bullet has some issues. Let me investigate further late today.

OK. Let me know if you have any progress on that.. Thanks!

Other than that, can I try anything else to improve the speed of my simulation? Changing to a different LCP solver would make any difference or not?

jslee02, 2018-10-24:
Practically `PgsBoxedLcpSolver` with low [iteration numbers](https://github.com/dartsim/dart/blob/release-6.7/dart/constraint/PgsBoxedLcpSolver.hpp#L55) could give you less accurate but faster results, but the current implementation is experimental (and not fully optimized yet) so no guarantee for now. But you could try it.

costashatz, 2018-10-24:
> Practically `PgsBoxedLcpSolver` with low [iteration numbers](https://github.com/dartsim/dart/blob/release-6.7/dart/constraint/PgsBoxedLcpSolver.hpp#L55) could give you less accurate but faster [...]

It is slower than the default configuration even if I decrease the maxIterations to 5 (at least in my setup). Just letting you know.. Thanks for your feedback. Let me know if you any update on Bullet collision..

costashatz, 2018-10-31:
@jslee02 Did you find anything regarding to why Bullet is not behaving as supposed to be with self-collisions? I tried to investigate it, but didn't find anything. Thanks..

jslee02, 2018-11-04:
@costashatz Sorry for the late response. I haven't tested this yet due to the hectic schedule of this quarter. I thought I could have a chance to take a look at this at the day, but couldn't.

Would you mind if you provide a simple test that Bullet doesn't work for self-collision? It would be much easier to investigate the issue for me. Thanks!