dartsim/dart #1187: Install DART on windows

EngineerZhang, 2018-11-02:
I currently use the vcpkg to download Dependencies and download the DART Source code,but I don't know where to put the downloaded dependencies？

jslee02, 2018-11-10:
It should be managed by `vcpkg`. One thing you should do is to set the cmake variable `CMAKE_TOOLCHAIN_FILE` to the `vcpkg.cmake` file. The command would be like:
```shell
cmake .. -DCMAKE_TOOLCHAIN_FILE=C:\tools\vcpkg\scripts\buildsystems\vcpkg.cmake  # the path you installed vcpkg
```

closed 2018-11-10 by jslee02

EngineerZhang, 2018-11-12:
Thanks for your replay!
I run the cmake as this command `cmake -DCMAKE_TOOLCHAIN_FILE=C:\Users\LittleZhang\Desktop\vcpkg\scripts\buildsystems\vcpkg.cmake -S C:\Users\LittleZhang\Desktop\dart-6.6.2\dart-6.6.2 -B C:\Users\LittleZhang\Desktop\dart-6.6.2\dart-6.6.2\build`，

but I got wrong message：

`--
-- ============================================
--                 DART 6.6.2
-- ============================================
CMake Error at C:/Users/LittleZhang/Desktop/cmake-3.13.0-rc3-win64-x64/cmake-3.13.0-rc3-win64-x64/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find EIGEN3 (missing: EIGEN3_INCLUDE_DIRS) (Required is at least
  version "3.0.5")
Call Stack (most recent call first):
  C:/Users/LittleZhang/Desktop/cmake-3.13.0-rc3-win64-x64/cmake-3.13.0-rc3-win64-x64/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindEIGEN3.cmake:32 (find_package_handle_standard_args)
  cmake/DARTFindDependencies.cmake:13 (find_package)
  CMakeLists.txt:136 (include)

-- Configuring incomplete, errors occurred!
See also "C:/Users/LittleZhang/Desktop/dart-6.6.2/dart-6.6.2/build/CMakeFiles/CMakeOutput.log".`

How can I do for this？Thans again！

jslee02, 2018-11-12:
It seems you generated 64-bit VS solution. Have you installed the dependencies in x64? It would be something like:
```
vcpkg install --recurse --triplet x64-windows assimp boost-system boost-filesystem ccd eigen3 fcl
```

EngineerZhang, 2018-11-13:
When I type the above command，I got these information：

`The following packages are already installed:
    assimp[core]:x64-windows
    boost-filesystem[core]:x64-windows
    boost-system[core]:x64-windows
    ccd[core]:x64-windows
    eigen3[core]:x64-windows
    fcl[core]:x64-windows
Starting package 1/6: eigen3:x64-windows
Package eigen3:x64-windows is already installed
Elapsed time for package eigen3:x64-windows: 133.1 us
Starting package 2/6: boost-filesystem:x64-windows
Package boost-filesystem:x64-windows is already installed
Elapsed time for package boost-filesystem:x64-windows: 115.1 us
Starting package 3/6: assimp:x64-windows
Package assimp:x64-windows is already installed
Elapsed time for package assimp:x64-windows: 115.4 us
Starting package 4/6: boost-system:x64-windows
Package boost-system:x64-windows is already installed
Elapsed time for package boost-system:x64-windows: 113.6 us
Starting package 5/6: ccd:x64-windows
Package ccd:x64-windows is already installed
Elapsed time for package ccd:x64-windows: 113.9 us
Starting package 6/6: fcl:x64-windows
Package fcl:x64-windows is already installed
Elapsed time for package fcl:x64-windows: 112.7 us

Total elapsed time: 1.547 ms

The package eigen3:x64-windows provides CMake targets:

    find_package(Eigen3 CONFIG REQUIRED)
    target_link_libraries(main PRIVATE Eigen3::Eigen)

The package ccd:x64-windows provides CMake targets:

    find_package(ccd CONFIG REQUIRED)
    target_link_libraries(main PRIVATE ccd)

The package fcl:x64-windows provides CMake targets:

    find_package(fcl CONFIG REQUIRED)
    target_link_libraries(main PRIVATE fcl)`

So，I am sure I have installed the dependecies. 
Then I add this sectence to the CMakeLists.txt of the dart

> SET(CMAKE_TOOLCHAIN_FILE "C:/Users/LittleZhang/Desktop/vcpkg/scripts/buildsystems/vcpkg.cmake")

Then I configure, got following messages:

============================================
                DART 6.6.2
============================================
Looking for octomap - NOT found, to use VoxelGridShape, please install octomap
Looking for doxygen - NOT found, to use generating API documentation, please install doxygen
Looking for ipopt - NOT found, to use dart-optimizer-ipopt, please install ipopt
Could NOT find OpenSceneGraph nor OpenThreads -- we will skip dart-gui-osg
If you believe you do have both OSG and OpenThreads installed, try setting OSG_DIR
Adding 11 components including the default component 'dart'
Adding 42 tests (comprehensive: 15, regression: 4, unit: 23)
CMake Warning at C:/Users/LittleZhang/Desktop/vcpkg/scripts/buildsystems/vcpkg.cmake:247 (_find_package):
  By not providing "FindTinyDNN.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TinyDNN", but
  CMake did not find one.

  Could not find a package configuration file provided by "TinyDNN" with any
  of the following names:

[... 111 lines cut ...]
  Value of lcpsolver_headers contained a newline; truncating

CMake Warning:
  Value of math_headers contained a newline; truncating

CMake Warning:
  Value of optimizer_headers contained a newline; truncating

CMake Warning:
  Value of optimizer_nlopt_headers contained a newline; truncating

CMake Warning:
  Value of planning_headers contained a newline; truncating

CMake Warning:
  Value of simulation_headers contained a newline; truncating

CMake Warning:
  Value of utils_headers contained a newline; truncating

CMake Warning:
  Value of utils_sdf_headers contained a newline; truncating

CMake Warning:
  Value of utils_urdf_headers contained a newline; truncating