dartsim/dart #1190: Ubuntu 16.04 (Xenial) libdart6-dev packages dependencies

karriforesthermit, 2018-11-08:
### Bug Report
- [x] I checked the [documentation](http://dartsim.github.io/) and found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
* libdart6-dev
* Ubuntu 16.04 (Xenial)
* repository packages ppa:dartsim.ppa

#### Expected Behavior
install as advertised at http://dartsim.github.io/install_dart_on_ubuntu.html

#### Current Behavior

Failure because of unmet dependencies

#### Steps to Reproduce

1.  Attempt to install 

$ sudo apt-get install libdart6-dev
[...]
The following packages have unmet dependencies:
 libdart6-dev : Depends: libdart6 (= 6.6.1-1734~201808060252~ubuntu16.04.1) but it is not going to be installed

2. Attempt to install dependencies

$ sudo apt-get install libdart6
[...]
The following packages have unmet dependencies:
 libdart6 : Depends: libboost-filesystem1.58.0 but it is not installable
            Depends: libboost-regex1.58.0 but it is not installable
            Depends: libboost-system1.58.0 but it is not installable
            Depends: libfcl0 but it is not installable
            Depends: liboctomap1.6v5 but it is not installable

3. Check what is available for current update

$ sudo apt-cache show libboost-filesystem1.\* | grep Package
Package: libboost-filesystem1.62.0
Package: libboost-filesystem1.62-dev
Package: libboost-filesystem1.63-dev
Package: libboost-filesystem1.63.0

And so on for other packages. 

Suspect that dependencies met by >= 1.58, not just == 1.58?

Thanks

jslee02, 2018-11-10:
Please correct me if I'm wrong. I guess this is because the dart packages are built with boost 1.58 binaries, and it's doesn't allow to link to higher versions of boost. I think in this case you should build DART from source rather than installing from the PPA.

jslee02, 2018-11-11:
Closing this issue. Please reopen if the problem is persistent!

closed 2018-11-11 by jslee02