dartsim/dart #1198: [suggestion] Please consider emphasizing the project name

yurivict, 2018-11-20:
Hi,

Please note that package for this project is sometimes called ```dartsim``` (ex. https://repology.org/metapackage/dartsim/versions), and is sometimes called ```libdart``` (ex. https://repology.org/metapackage/libdart/versions). You call it ```dart```, but there is also the programming language called ```dart```, see here https://repology.org/metapackage/dart/versions .

You may want to emphasize here in the README what the project name is. I suggest you choose something unique, either ```dartsim``` or ```libdart``` or maybe ```dart-robotics```.

I would like to create the FreeBSD port but I am in doubt what the port's name should be.

Your ```include/dart``` headers location and ```share/doc/dart``` locations might conflict with the unrelated ```dart``` project.

Thank you,
Yuri

acxz, 2019-02-17:
Hello, have you taken a look at the following issue: https://github.com/dartsim/dart/issues/479? It might help you choose a name for the FreeBSD port.

jslee02, 2019-02-18:
The project name is DART, but we use `dartsim` as the package name when there is name confliction in the namespace of the package manager. For example, the [Homebrew package](https://github.com/Homebrew/homebrew-core/blob/master/Formula/dartsim.rb) and [ROS package](https://github.com/dartsim/ros-dartsim-release). One exception is that `libdart`, which is the Debian package name. The naming convention is `lib<name>` when it's a library so `libdart`.

I would recommend using `dartsim` for the FreeBSD port.

mxgrey, 2019-02-18:
We may want to consider using `libdartsim` for the library name as well for consistency's sake. And also change the cmake config to work like `find_package(dartsim ...)`. But those changes probably shouldn't be made until the next major release (7).

yurivict, 2019-02-18:
@jslee02 I followed your recommendation, thanks: https://www.freshports.org/misc/dartsim/

jslee02, 2019-02-18:
Awesome! Thanks for having DART in FreeBSD port!

yurivict, 2019-02-18:
You're welcome!

jslee02, 2019-07-14:
Closing this issue as there is nothing left to be addressed. Please reopen if I'm missing something here.

closed 2019-07-14 by jslee02