dartsim/dart #1199: Please install examles into a dedicated directory

yurivict, 2018-11-20:
You install examples into ```share/doc/dart/examples``` while examples are commonly installed into their own directory. For example, on FreeBSD port docs by default go into ```share/doc/{project-name}``` and exampled go into ```share/examples/{project-name}```. Users can individually choose if they would like to have examples and docs installed with port options ```EXAMPLES``` and ```DOCS```. It is better for users this way.

jslee02, 2019-04-24:
Is this `FreeBSD` specific convention? Because it seems the pattern isn't being used in Ubuntu since `share/examples/` does not exist by default.

yurivict, 2019-04-24:
Then you should have a cmake variable for this directory, so that patching wouldn't be needed.