dartsim/dart #1200: Example Compile Error in Mac OS Mojave

shbang91, 2018-11-29:
Hello,
I am working on Mac OS Mojave with DART 6.5.0
I just go to the
```
examples/atlasSimbicon
mkdir build & cd build
cmake .. & make
```
and getting the following error
```
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [atlasSimbicon] Error 1
make[1]: *** [CMakeFiles/atlasSimbicon.dir/all] Error 2
make: *** [all] Error 2
```
I think this issue is related to the version of Mac OS.
Do you have a plan to fix this?

shbang91, 2018-11-29:
Thanks a lot!

scpeters, 2018-11-29:
actually, this looks related to https://github.com/ros/console_bridge/issues/52

scpeters, 2018-12-12:
I think this will be fixed by https://github.com/ros/console_bridge/pull/60

jslee02, 2019-01-06:
@scpeters Thanks for the fix! However, the fix didn't work for me. I think it fails to load `console_bridge` (and even `console_bridge::console_bridge`) because the target is not imported by just loading `DARTConfig.cmake`. Instead, `DARTConfig.cmake` only store the target name in `DART_LIBRARIES` without importing the target actually. I hope #1209 fixes this issue.

scpeters, 2019-01-09:
Thanks @jslee02; I thought the target namespace was the problem, but I guess it's an issue in dart. Yes, #1209 seems to fix it.

jslee02, 2019-01-09:
@scpeters Thanks for the confirmation. I think I learned a lot about target including target namespace!

Closing this issue as it's resolved. @shbang91 Please let us know if this is still persistent for you.

closed 2019-01-09 by jslee02

scpeters, 2019-01-09:
It's resolved on the `release-6.7` branch, but it won't be fully resolved for downstream users until it's released in homebrew. Do you have a planned date for the 6.7 release?

jslee02, 2019-01-09:
Sure. I plan to release 6.7 once #1212 is merged.

PR #1209 (2019-01-08): Import dependency targets when loading DARTConfig.cmake
Currently, the transitive dependencies of DART are propagated by using CMake variables. For example, all the dependency libraries are stored in `DART_LIBRARIES`, which will be defined by loading `DARTConfig.cmake` through calling `find_package(DART)` in a downstream project. 

However, this approach can fail to load the dependency. The items in `DART_LIBRARIES` can be either of the path to a library file or a target. Path items work fine (as long as the path is correct), but not targets because the downstream project don't know how to imp [... cut]
commit ef20d1a7 (2019-01-08): Import dependency targets when loading DARTConfig.cmake (#1209)

Currently, the transitive dependencies of DART are propagated by using CMake variables. For example, all the dependency libraries are stored in `DART_LIBRARIES`, which will be defined by loading `DARTConfig.cmake` through calling `find_package(DART)` in a downstream project. 

However, this approach can fail to load the dependency. The items in `DART_LIBRARIES` can be either of the path to a library file or a target. Path items work fine (as long as the path is correct), but not targets because the downstream project don't know h [... cut]